wb2osz/direwolf #254: Empty callsigns in path

srcejon, 2020-02-28:
Packets received from AISAT appear to have empty callsigns in the path:

[rx>ig] AISAT-1>CQ,,,qAO,M7RCE::CQ-0 :From AMSAT INDIA & Exseed Space |3179768|38|36|1070{482

Which are then dropped by aprsc igate software, with the error: invalid callsign in path. 

Is this a direwolf problem, or the a problem in the structure of the received packets from AISAT?

srcejon, 2020-02-28:
I can reproduce this by creating a packet where a via path is "       ". Direwolf just outputs ,,. Perhaps it would be better if it stripped it?

srcejon, 2020-02-28:
Attached is a possible patch to ax25_pad.c to workaround this:

[ax25_pad_patch.txt](https://github.com/wb2osz/direwolf/files/4266905/ax25_pad_patch.txt)

wb2osz, 2020-02-28:
The AISAT software was written by someone who did not understand the APRS protocol.  The satellite was obviously launched without testing the APRS software first to see if it was actually compatible with the rest of the world.  It has multiple problems. 

Numerous software developers and countless IGate operators should not be expected to kludge the entire worldwide APRS-IS network to accommodate one faulty implementation of a simple protocol that has been around for decades.   That would only encourage others to be careless and not do any testing.

 Some of the discussion can be found here:

http://lists.tapr.org/pipermail/aprssig_lists.tapr.org/2019-April/047964.html 
http://lists.tapr.org/pipermail/aprssig_lists.tapr.org/2019-April/047994.html
http://lists.tapr.org/pipermail/aprssig_lists.tapr.org/2019-April/047995.html
http://lists.tapr.org/pipermail/aprssig_lists.tapr.org/2019-April/047996.html
http://lists.tapr.org/pipermail/aprssig_lists.tapr.org/2019-April/047997.html

closed 2020-02-28 by wb2osz