wb2osz/direwolf #256: Feature Request: Allow pseudo terminal option (-p) to specify symlink name

radar27, 2020-03-05:
Greetings,

In some cases, multiple instances of Direwolf are running, but need support for for a Virtual KISS TNC as provided by the -p startup option.  By default this is always linked to /tmp/kisstnc, creating a race condition on the symlink by whichever instance starts last.  

Would it be possible for -p to accept an optional argument to specify the name of the target symlink?  Something like:

> direwolf -c direwolf-40m.conf -p /tmp/kiss40m
> direwolf -c direwolf-2m.conf -p /tmp/kiss2m
> direwolf -c direwolf-70cm.conf -p /tmp/kiss70cm

From there one could attach the 40m instance via /tmp/kiss40m, the 2m instance via /tmp/kiss2m and the 70cm instance via /tmp/kiss70cm.  If the -p option is specified with no argument, it continue to default to /tmp/kisstnc as to not disrupt backwards compatibility.

73!

dranch, 2020-03-07:
This seems like a useful enhancement to me for those who wish to use the native Linux AX.25 stack with one radio per Direwolf instance

n9seo, 2021-04-25:
Please can we add this?   I'd almost rather set up the devnames with 'socat' than let DW do it and if we can send an arg to '-p' that would solve this.  Bonus for allowing more than one -p device name for those running dual stack things in linux like ax25 and JNOS or something else.    Since we cant have two things consume a psuedo serial port that should make sense.