wb2osz/direwolf #262: Feature request: Please add compatibility with Navit

Supermagnum, 2020-04-04:
Navit is a free and open-source, modular, touch screen friendly, car navigation system with GPS tracking, realtime routing engine and support for various vector map formats. It features both a 2D and 3D view of map data.

Navit supports a variety of operating systems and hardware platforms.
It is currently not compatible with APRS.

Navits official web page:
https://www.navit-project.org/

Github page:
https://github.com/navit-gps

Navit Documentation:
http://doxygen.navit-project.org/

Link to the suggestion( patch request): 
https://github.com/navit-gps/navit/issues/982

wb2osz, 2020-04-05:
direwolf already generates these types of sentences for use with GPS displays or other mapping systems:

 *	$GPWPL		- NMEA generic with only location and name.
 *	$PGRMW		- Garmin, adds altitude, symbol, and comment
 *					  to previously named waypoint.
 *	$PMGNWPL	- Magellan, more complete for stationary objects.
 *	$PKWDWPL	- Kenwood with APRS style symbol but missing comment.
 
The last one would be best for use with APRS.

First you would need to convince the Navit authors to accept objects, from another application, in these or some other format.

direwolf currently uses a serial port for compatibility with older existing equipment.  Using UDP would probably make more sense when using two applications.

wb2osz, 2020-04-23:
First you would need to convince the Navit authors to accept objects, from another application, in one of these or some other standard format such as KML.  After that we can discuss how to send APRS Position Reports, Objects, etc. to Navit.

I think that talking about APRS would only confuse the Navit developers.  It would be better to describe the desired functionality in more general terms such as a mechanism to display stationary and moving objects provided real-time by some other application.

wb2osz, 2020-12-31:
No response.
Reopen after you convince Navit authors to accept moving objects, from another application.

Supermagnum, 2020-12-31:
https://github.com/navit-gps/navit/issues/982
I don't think I can do anything more :(

wb2osz, 2020-12-31:
This looks interesting https://github.com/N2ZLC/dire_wolf_to_navit
I was thinking more in terms of a network interface to feed in moving objects, but a file interface would work too.
Maybe if we can get a conversation going, with all the hams interested in this topic, we can come up with something.
I think there would be a lot of value to this. For example, you might be chasing a descending balloon and use the navigation system to guide you to its most recent changing position. The normal APRS mapping applications are not suited for that,

wb2osz, 2020-12-31:
How about starting a new conversation here  https://groups.io/g/direwolf  to get some momentum going on this?

Supermagnum, 2023-03-26:
Any progress on this, or has it stalled completely?

Supermagnum, 2023-04-01:
> First you would need to convince the Navit authors to accept objects, from another application, in one of those mentioned or some other standard format such as KML. After that we can discuss how to send APRS Position Reports, Objects, etc. to Navit.
> 
> I think that talking about APRS would only confuse the Navit developers. It would be better to describe the desired functionality in more general terms such as a mechanism to display stationary and moving objects provided real-time by some other application.
> 
> After Navit has the capability to display fixed and moveable objects from some other application, the direwolf part can begin.

https://github.com/navit-gps/navit/issues/1231