wb2osz/direwolf #263: Ignoric SLOTTIME and PERSIST setting

kandev, 2020-04-05:
I'm using Direwolf 1.5 on Raspberry PI 3 with Ubuntu 18.04
All seems to be working fine, except that I can't insert any delay between digipeating packets. Basically digipeating start immediately after packet is recieved.
Currently I have this in my configuration:
```
PTT GPIO 25
DWAIT 0
SLOTTIME 100
PERSIST 5
TXDELAY 30
TXTAIL 10
```
This should give very little chance to transmit at earliest, 1 second after the packet is received. But in reality PTT/digipeating is triggered immediately after packet enters.
Am I missing something or what could be wrong?! Any ideas?

dranch, 2020-04-05:
None of those values are intended to delay the digipeating function.  They are for just interfacing with your specific radio's key up characteristics.  The digipeater section has the "DEDUP" feature which can minimize duplication and there is also the "satgate" feature offers a delay as well but I don't think that's appropriate here.  Please read the UserGuide for more details on these items as well as the "“viscous” discussion which is supported by APRX but not Direwolf.

kandev, 2020-04-05:
Thanks for clearing this out!  I guess I have missed it if it's written somewhere in the manual.

wb2osz, 2020-04-05:
As David stated, APRS digipeaters are supposed to retransmit immediately, after the channel is clear, rather than waiting the usual random time.  This is intentional.

closed 2020-04-05 by wb2osz