wb2osz/direwolf #265: Feature request: emit PTT on multiple gpio pins

craigerl, 2020-04-12:
PTT 24 26

I tried that and only 24 goes high on ptt.  would be fun to hookup a red LED to 26.

mattbk, 2021-04-12:
As a workaround, could you put the LED in series with the PTT circuit?

craigerl, 2021-04-12:
Possibly, but I can't guarantee every radio will like the voltage drop over the diode (may not engage transmitter),
nor can I guarantee the size of the current limiting resistor in every radio (might be dim, smoke, or not turn on at all).

The LEDs on this video aren't possible with the current direwolf,

https://www.youtube.com/watch?v=4JNWIGvlh64

This software gpio-mirror.py solution works 100%, but burns cpu and memory... both scarce on the Pi Zero.

```
#!/usr/bin/python3

import pyinotify
import RPi.GPIO as GPIO

def handle_change(cb):
   with open('/sys/class/gpio/gpio12/value', 'r') as f:
      #print(f.read())
      status = f.read(1)
      if status == '0':
         #print("OFF")
         GPIO.output(26, GPIO.LOW)
         #system("echo 16 > /sys/class/gpio/export")   # re-enable green led, hangs direwatch
      else:
         #print("ON")
         GPIO.output(26, GPIO.HIGH)
         #GPIO.output(16, GPIO.LOW)                       # turn off green led
         #system("echo 16 > /sys/class/gpio/unexport") # disable green led, hangs direwatch
   f.close

def null_function(junk):  # default callback prints tons of debugging info
   return()

GPIO.setmode(GPIO.BCM)    # logical pin numbers, not BOARD
GPIO.setup(26, GPIO.OUT)  #  RED
#GPIO.setup(16, GPIO.OUT)  #  GREEN  (mute this in case of cross talk)
GPIO.setwarnings(False)   # suppress pin-is-in-use warning

# Instanciate a new WatchManager (will be used to store watches).
wm = pyinotify.WatchManager()

# Associate this WatchManager with a Notifier
notifier = pyinotify.Notifier(wm, default_proc_fun=null_function)

# Add a new watch
wm.add_watch('/sys/class/gpio/gpio12/value', pyinotify.IN_MODIFY)

# Loop forever and handle events.
notifier.loop(callback=handle_change)

```

ab0tj, 2021-04-12:
Most radios have their PTT pin pulled to ground to transmit. So why not just hook up the LED with the cathode on the PTT output? Then it both acts as a pullup and a PTT LED.

dranch, 2021-04-12:
It sounds like there might be better ways to poll GPIO pins with lower CPU load:

   https://www.raspberrypi.org/forums/viewtopic.php?t=167807

Other options found via using the search "raspberry pi low cpu load method to monitor a GPIO pin"

craigerl, 2021-04-12:
Nope, I'm using the kernel's inotify service for a state change... not polling, so cpu isn't really bad, it's mostly the overhead of yet-another-python-interpretery-loaded-into-memory just to monitor the direwolf ptt pin and mirror its inverse on another pin to drive a led.

mattbk, 2021-04-13:
Another idea might be to set up another channel with duplicate settings except for the PTT line.

CtrlC-Root, 2024-06-23:
I implemented a work around for this [detailed here](https://github.com/wb2osz/direwolf/issues/61#issuecomment-2185354576). If you have a spare serial port on your system you could configure Direwolf to use the RTS signal on that (`PTT /dev/ttyANY RTS` I believe) and then toggle multiple GPIO pins via the ttyhook script. As far as I can tell from a cursory glance at the Direwolf code (1) it opens the serial port in read only mode and (2) it only ever toggles the RTS or DTR signals on it so it should be safe to use with any serial port even one used by other software for other purposes.