wb2osz/direwolf #270: Direwolf terminal blinking.

F4FVI, 2020-05-03:
Hello,
I am actually creating a Digipeater using Direwolf on a Raspberry Pi 2.
Once Direwolf is installed, when the terminal starts, Direwolf display is blinking. 

Here is how I have installed Direwolf on my RaspberryPi2:

1) I flashed Raspbian (Raspbian Buster with desktop) on the SD Card.

2) Then I open a terminal and I did **sudo apt-get update** and **sudo apt-get upgrade**.

3) To install Direwolf, I I have used the following command in a terminal:

**cd ~  
git clone https://github.com/wb2osz/direwolf  
cd direwolf
make
sudo make install
make install-conf** 

4) After, when I click on the shortcut to start Direwold, the terminal is blinking.
I have the same result blinking like shown here [https://github.com/wb2osz/direwolf/issues/215]

5) The only way found to stop this blinking problem, is to start Direwolf in black and white following theses steps:

1. Creating a shortcut of Dire Wolf icon on the desktop

2. Creating a file **direwolf.sh** in /home/pi with **-t 0** at the end:

**#!/bin/bash
cd /home/pi/Desktop
lxterminal -e direwolf _-t 0_**

3. Creating a repertory **autostart** in /home/pi/.config

4. In the autostart repertory, creating a file named **direwolf.desktop** with in it:

**[Desktop Entry]
Type=Application
Name=Journal
Exec=lxterminal -e ./direwolf.sh
Name[fr_FR]=direwolf**

I have read the issue subject posted here before, but no solution given actually works to solve the problem. It would be great to have the Direwolf terminal in colour withtout blinking.

Could anyone help me to solve this problem? 
Thank you very much in advance;

dranch, 2020-05-03:
The choices in color are controlled by the src/textcolor.c file.  Read the top of that file's comments to get a good understanding of what's the intent, the found issues with various terminals, etc.    If you read the Direwolf UserGuide.pdf file (PDF page 141), it mentions you can try manually running Direwolf with "-t 9" to see alternative color options.  then try one that might work better for you using "-t ?" where you replace ? with your new color map choice.

F4FVI, 2020-05-03:
Thank you very much for you answer.

I read the textcolor.c file and the user guide as well.

Unthanksfully, I have tried to replace "**-t ?**" with ?= 1,2,3,4,5,6,7,8,9 and none of them stop blinking. 

Only the "**-t 0"** in black and white color does not blink. 

Thank you anyway for your answer. Maybe somebody else found a solution or maybe we have to wait a newer version of Direwolf.

dranch, 2020-05-03:
What version of Direwolf are you using?  Check out Direwolf v1.6 Beta in the "develop" branch from the Git repo.  There are new color mapping options in there for say LXterminal, etc.

F4FVI, 2020-05-03:
I am actually using the version 1.5 of DireWolf.

I have to say that I am not a linux user so I mainly copy-paste the tutorials.

Could you please say me by which link do I have to replace the installation code to install the Version 1.6 Beta?

I guess it is different than **_https://github.com/wb2osz/direwolf_**

cd ~
git clone **_https://github.com/wb2osz/direwolf_**
cd direwolf
make
sudo make install
make install-conf

dranch, 2020-05-03:
Not a Linux thing.. it's a Git and branch thing:
--
cd ~
git clone https://github.com/wb2osz/direwolf
cd direwolf
git checkout dev
.
.
--
(continue with your previous list of commands}

F4FVI, 2020-05-03:
Thank you.

I did it but it looks like that from the the make **command** nothing works.

Did I make something wrong?

pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ git clone https://www.github.com/wb2osz/direwolf
Clonage dans 'direwolf'...
warning: redirection vers https://github.com/wb2osz/direwolf.git/
remote: Enumerating objects: 22, done.
remote: Counting objects: 100% (22/22), done.
remote: Compressing objects: 100% (19/19), done.
Réception d'objets:  16% (408/2499), 908.01 KiB | 894.00 KiB/sRéception d'objets:  16% (408/2499), 944.01 KiB | 238.00 KiB/sRéception d'objets:  16% (409/2499), 944.01 KiB | 238.00 KiB/sRéception d'objets:  16% (410/2499), 2.39 MiB | 494.00 KiB/s  Réception d'objets:  16% (411/2499), 3.83 MiB | 604.00 KiB/s  Réception d'objets:  16% (413/2499), 4.32 MiB | 631.00 KiB/s  Réception d'objets:  16% (414/2499), 5.28 MiB | 641.00 KiB/s  Réception d'objets:  16% (415/2499), 6.15 MiB | 692.00 KiB/s  Réception d'objets:  16% (418/2499), 6.15 MiB | 692.00 KiB/s  Réception d'objets:  16% (422/2499), 9.29 MiB | 968.00 KiB/s  Réception d'objets:  16% (424/2499), 9.78 MiB | 968.00 KiB/s  Réception d'objets:  17% (425/2499), 9.78 MiB | 968.00 KiB/s  Réception d'objets:  17% (428/2499), 11.09 MiB | 964.00 KiB/s Réception d'objets:  17% (429/2499), 11.61 MiB | 969.00 KiB/s Réception d'objets:  17% (430/2499), 13.04 MiB | 967.00 KiB/s Réception d'objets:  17% (430/2499), 13.99 MiB | 866.00 KiB/s Réception d'objets:  17% (431/2499), 15.02 MiB | 967.00 KiB/s Réception d'objets:  17% (431/2499), 15.99 MiB | 975.00 KiB/s Réception d'objets:  17% (431/2499), 16.94 MiB | 966.00 KiB/s Réception d'objets:  17% (432/2499), 17.91 MiB | 967.00 KiB/s Réception d'objets:  24% (600/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  24% (604/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  25% (625/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  26% (650/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  27% (675/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  28% (700/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  29% (725/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  30% (750/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  31% (775/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  32% (800/2499), 19.35 MiB | 966.00 KiB/s Réception d'objets:  33% (825/2499), 19.82 MiB | 966.00 KiB/s Réception d'objets:  34% (850/2499), 19.82 MiB | 966.00 KiB/s Réception d'objets:  35% (875/2499), 19.82 MiB | 966.00 KiB/s Réception d'objets:  36% (900/2499), 19.82 MiB | 966.00 KiB/s Réception d'objets:  37% (925/2499), 19.82 MiB | 966.00 KiB/s Réception d'objets:  38% (950/2499), 19.82 MiB | 966.00 KiB/s Réception d'objets:  39% (975/2499), 19.82 MiB | 966.00 KiB/s Réception d'objets:  39% (975/2499), 20.79 MiB | 967.00 KiB/s Réception d'objets:  39% (976/2499), 21.26 MiB | 967.00 KiB/s Réception d'objets:  39% (976/2499), 22.69 MiB | 965.00 KiB/s Réception d'objets:  39% (976/2499), 23.66 MiB | 970.00 KiB/s Réception d'objets:  39% (976/2499), 24.60 MiB | 968.00 KiB/s Réception d'objets:  39% (977/2499), 25.04 MiB | 711.00 KiB/s Réception d'objets:  39% (998/2499), 27.04 MiB | 967.00 KiB/s Réception d'objets:  39% (999/2499), 27.98 MiB | 967.00 KiB/s Réception d'objets:  40% (1000/2499), 27.98 MiB | 967.00 KiB/sRéception d'objets:  40% (1000/2499), 28.94 MiB | 967.00 KiB/sRéception d'objets:  40% (1000/2499), 29.90 MiB | 969.00 KiB/sRéception d'objets:  40% (1015/2499), 30.38 MiB | 968.00 KiB/sRéception d'objets:  41% (1025/2499), 30.78 MiB | 1.14 MiB/s  Réception d'objets:  42% (1050/2499), 30.78 MiB | 1.14 MiB/s  Réception d'objets:  43% (1075/2499), 30.78 MiB | 1.14 MiB/s  Réception d'objets:  44% (1100/2499), 30.78 MiB | 1.14 MiB/s  Réception d'objets:  44% (1109/2499), 30.78 MiB | 1.14 MiB/s  Réception d'objets:  44% (1110/2499), 32.77 MiB | 966.00 KiB/sRéception d'objets:  44% (1112/2499), 33.72 MiB | 965.00 KiB/sRéception d'objets:  44% (1112/2499), 34.68 MiB | 964.00 KiB/sRéception d'objets:  45% (1125/2499), 34.68 MiB | 964.00 KiB/sRéception d'objets:  46% (1150/2499), 34.68 MiB | 964.00 KiB/sRéception d'objets:  47% (1175/2499), 34.68 MiB | 964.00 KiB/sRéception d'objets:  47% (1178/2499), 35.63 MiB | 1.06 MiB/s  Réception d'objets:  47% (1179/2499), 36.59 MiB | 968.00 KiB/sRéception d'objets:  47% (1183/2499), 37.55 MiB | 969.00 KiB/sRéception d'objets:  47% (1183/2499), 38.49 MiB | 964.00 KiB/sRéception d'objets:  47% (1184/2499), 38.97 MiB | 969.00 KiB/sRéception d'objets:  48% (1200/2499), 38.97 MiB | 969.00 KiB/sRéception d'objets:  49% (1225/2499), 39.45 MiB | 969.00 KiB/sRéception d'objets:  50% (1250/2499), 39.45 MiB | 969.00 KiB/sRéception d'objets:  50% (1259/2499), 40.41 MiB | 968.00 KiB/sRéception d'objets:  50% (1262/2499), 41.37 MiB | 969.00 KiB/sRéception d'objets:  50% (1263/2499), 42.31 MiB | 964.00 KiB/sRéception d'objets:  50% (1263/2499), 42.65 MiB | 854.00 KiB/sRéception d'objets:  50% (1266/2499), 43.68 MiB | 966.00 KiB/sRéception d'objets:  50% (1266/2499), 45.11 MiB | 965.00 KiB/sRéception d'objets:  50% (1266/2499), 46.25 MiB | 964.00 KiB/sRéception d'objets:  50% (1267/2499), 46.72 MiB | 963.00 KiB/sRéception d'objets:  50% (1267/2499), 47.54 MiB | 966.00 KiB/sRéception d'objets:  50% (1269/2499), 48.57 MiB | 965.00 KiB/sRéception d'objets:  51% (1275/2499), 48.57 MiB | 965.00 KiB/sRéception d'objets:  52% (1300/2499), 48.57 MiB | 965.00 KiB/sRéception d'objets:  53% (1325/2499), 48.57 MiB | 965.00 KiB/sRéception d'objets:  53% (1342/2499), 49 [... cut ...]
Extraction des fichiers: 100% (502/502), fait.
La branche 'dev' est paramétrée pour suivre la branche distante 'dev' depuis 'origin'.
Basculement sur la nouvelle branche 'dev'
pi@raspberrypi:~/direwolf $ make
make: *** Pas de cible spécifiée et aucun makefile n'a été trouvé. Arrêt.
pi@raspberrypi:~/direwolf $ sudo make install
make: *** Aucune règle pour fabriquer la cible « install ». Arrêt.
pi@raspberrypi:~/direwolf $ make install-conf
make: *** Aucune règle pour fabriquer la cible « install-conf ». Arrêt.
pi@raspberrypi:~/direwolf $

dranch, 2020-05-03:
In the new Direwolf branch, you aren't reading the new README.md or UserGuide PDF file as the build process has changed.  Read those on how to get Cmake going, etc.   The new documentation is a work in progress and I've added a few more details in my doc as well:

http://www.trinityos.com/HAM/CentosDigitalModes/RPi/rpi2-setup.html#24.compile-direwolf

F4FVI, 2020-05-03:
Thank you very much, I will have a look little by little because it starts to be really complicated for non initiated to Raspbian/Linux guy like me ;)

Thank you for your help anyway.

F4FVI, 2020-05-04:
I found a solution.

It is necessary to change the terminal emulator.
Instead of LXTERMINAL I have chosen TERMINOLOGY.

The color are not blinking anymore but... Clear grey background with clear green letter make not readable the text unthanksfully.

Thank you for your help anyway. 
Greetings

wb2osz, 2020-05-04:
The solution is to use the version in the "dev" branch.
Soon this should be in the stable release 1.6.

closed 2020-05-04 by wb2osz

wb2osz, 2020-12-31:
Release 1.6 has been out for a while now.
Use "-t 9" on the command line to see the effect of different numbers.
Choose the best one for your terminal type.