wb2osz/direwolf #275: aborted connects via agwpe port don't abort

pe1rrr, 2020-05-22:
Hello,

When connecting to a station, then having 2nd thoughts and disconnecting (sending 'd' before the remote station has responded) dw (direwolf) continues to attempt connections until it reaches the maximum retry limit. 

This means it is possible to detach an agwpe terminal session from dw without it first stopping the SABM polls. 

`[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
DCD 0 = 1
DCD 0 = 0
PTT 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
DCD 0 = 1
DCD 0 = 0
PTT 0 = 0
Disconnect from EI2GYB-3 ...
Disconnect from EI2GYB-3 ...
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
PTT 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
DCD 0 = 1

PE1RRR-13 audio level = 0(1/1)    _0_
[0.1] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
PTT 0 = 0
DCD 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
PTT 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
PTT 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
Disconnect from EI2GYB-3 ...
PTT 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)

PE1RRR-13 audio level = 0(0/0)    305
[0.1] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
PTT 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
PTT 0 = 0
PTT 0 = 1
[0L] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)

PE1RRR-13 audio level = 0(1/1)    30_
[0.1] PE1RRR-13>EI2GYB-3:(SABM cmd, p=1)
PTT 0 = 0
Failed to connect to EI2GYB-3 after 10 tries.`

The double reporting of disconnect state 'Disconnect from EI2GYB-3 ...' also seems a bit dubious.

Let me know if I need to provide anything else.

jmkristian, 2020-07-20:
I agree this is annoying. It would be nice to send DISC or DM, or at least stop retrying SABM.

lightspeed397, 2020-11-08:
I agree I would like to send a few commands and also send messages.

wb2osz, 2023-05-07:
This has been fixed in the dev branch, commit 8e32286.

closed 2023-05-07 by wb2osz