wb2osz/direwolf #276: Raspberry Pi, Direwolf, Digipeater, Igate, Soundcard version

rgrobinson131, 2020-05-26:
Hi All im trying to build the above following the guide https://godseyonline.net/raspberry-pi-direwolf-digipeater-igate-soundcard-version/

All workign apart from I cannot get the auto start to work.. when i try sudo nano /home/pi/.config/lxsession/LXDE-pi/autostart
it returns with [ Directory '/home/pi/.config/lxsession/LXDE-pi' does not exist ]

dranch, 2020-05-26:
We have no idea who's guide that is or if there are any errors in it.  It's recommend you follow the Direwolf User guide at https://github.com/wb2osz/direwolf/blob/master/doc/User-Guide.pdf .  If you have any questions, please join the Direwolf support group at direwolf@groups.io and post your question there.  Github issues are for real bugs or feature enhancements.  Please close this issue.

rgrobinson131, 2020-05-26:
ive used
chmod +x dw-start.sh

then opend crontab -e and added the line
***** /home/pi/dw-start.sh >/dev/null 2>&1

rebooted and nothing happens?

wb2osz, 2020-05-26:
This "issues" section is for reporting software defects ("bugs") or making enhancement requests.
For questions like this, please use the discussion forum.  https://groups.io/g/direwolf  
Are you missing spaces in the crontab entry?  There must be spaces between the "*" characters.

closed 2020-05-26 by wb2osz