wb2osz/direwolf #278: Data received from network KISS applcation isn't sent to APRS-IS

g0wfv, 2020-05-31:
As per the subject. Data goes out over radio (audio) device, but doesn't (reliably) get sent to the APRS-IS via the igate.  It was working, but stopped working - I can't get it to start again!  Same symptoms in 1.5 and 1.6 (pulled latest dev and recompiled)

> Attached to KISS TCP client application 1...
> 
> Ready to accept KISS TCP client application 0 on port 8001 ...
> 
> <<< Data frame from KISS client application, port 0, total length = 60
>   000:  c0 00 82 a0 ae b0 60 62 60 8e 60 ae 8c ac 40 60  ......`b`.`...@`
>   010:  ae 92 88 8a 62 40 63 03 f0 5f 30 35 33 31 31 35  ....b@c.._053115
>   020:  35 30 63 31 30 30 73 30 31 37 67 30 31 37 74 30  50c100s017g017t0
>   030:  37 31 68 33 30 62 31 30 32 34 30 c0              71h30b10240.
> 
> <<< Data frame from KISS client application, port 0, total length = 94
>   000:  c0 00 82 a0 ae b0 60 62 60 8e 60 ae 8c ac 40 60  ......`b`.`...@`
>   010:  ae 92 88 8a 62 40 63 03 f0 3e 57 65 61 74 68 65  ....b@c..>Weathe
>   020:  72 20 72 65 70 6f 72 74 20 64 65 72 69 76 65 64  r report derived
>   030:  20 66 72 6f 6d 3a 20 4d 45 54 41 52 20 45 47 58   from: METAR EGX
>   040:  57 20 33 31 31 35 35 30 5a 20 31 30 30 31 35 4b  W 311550Z 10015K
>   050:  54 20 32 32 2f 30 34 20 51 31 30 32 34 c0        T 22/04 Q1024.
> 
> KISS client application 1 has gone away.
> 
> [0L] G0WFV>APWX01,WIDE1-1:_05311550c100s017g017t071h30b10240
> [0L] G0WFV>APWX01,WIDE1-1:>Weather report derived from: METAR EGXW 311550Z 10015KT 22/04 Q1024
> [rx>ig] #
> [rx>ig] #

direwolf.conf ...

> ACHANNELS 1
> CHANNEL 0
> MYCALL G0WFV
> IGSERVER euro.aprs2.net
> IGLOGIN G0WFV 13716
> PBEACON sendto=IG delay=0:30 every=60:00 symbol="igate" overlay=R lat=53.176N long=0.555W comment="Waddington Rx iGate (Radio not connected!)"

toyotahead, 2021-04-13:
I am having the same issue. 

I also see you are having the KISS client application 1 has gone away as well.

dranch, 2021-04-13:
Are you sure the issue isn't with the APRS-IS infrastructure?  In g0wfv's example, he's using euro.aprs2.net which is a high level redirector many sub-level hosts:

host euro.aprs2.net
euro.aprs2.net has address 188.174.166.103
euro.aprs2.net has address 91.205.92.26
euro.aprs2.net has address 94.33.51.3
euro.aprs2.net has address 85.90.180.26
euro.aprs2.net has address 18.202.153.44
euro.aprs2.net has address 85.214.83.187
euro.aprs2.net has address 193.106.216.154
euro.aprs2.net has address 44.141.143.40

At the time of your failure, did you have an ESTABLISHED TCP connection to port 14580 on one of those hosts?  You can confirm this by running "netstat -an | grep 14580" on your Direwolf machine.  Once you know what APRS-IS server you're connected to, you can also get more stats by looking at http://<your connected APRS-IS server IP>:14501 .  At the top of this webpage, you can see the server admin and you can reach out to them to see if they are seeing errors from your station, etc.

toyotahead, 2021-04-13:
@dranch 

My DW setup is pointing to APRS-IS server: noam.aprs2.net and yes my connection to the specific north america server on the port 14501.

I should make sure this point is clear.  Received RF packets are being igate'd correctly to APRS-IS. It is the packets received from connected KISS TCP-IP clients. These packets when received over KISS/TCP-IP only go out over RF, and not also to the APRS-IS server. 

[image]

g0wfv, 2021-07-23:
This is still an issue in 1.7.