wb2osz/direwolf #279: [enhancement] Support for DCD LED over CM108

m1geo, 2020-06-02:
It would be very useful to have a DCD LED onto the radio interface board via a CM108 using one of the other GPIO pins beside GPIO3 (used for PTT).

wb9vxy, 2022-09-26:
There is a DCD LED on the PAUL board that is connected to a GPI04 pin on the CM108.
How can I link that the the DCD signal in Direwolf?

dranch, 2022-09-26:
It's not clear what a "PAUL" board is but assuming this is driven by a CM108 audio chip, I take it you're asking to manage the DCD LED in a similar way to how Direwolf supports the  PTT with C-Media CM108/CM119 GPIO pins as documented in the Direwolf User Guide Section 9.2.9.5.  Is that correct?  If that's correct please update this Github issue and WB2OSZ will decide if will be considered or not.  If you can contribute actual code to support this, that will go a long ways to get it officially added.

wb9vxy, 2022-09-26:
I believe that the CM108 functionality in Direwolf comes from hamlib.

I have requested the same feature on the hamlib github site

m1geo, 2022-09-26:
As an aside, the PCB routing on that board is _wild_! I enjoy the USB tracks! They're supposed to be differential pairs, yet near the IC they split around some reisistors and each have a stub on them to a capacitor that certainly isn't supposed to be there, and certainly shouldn't stub the line (but is likely there for EMC). Just made me smile.

wb2osz, 2022-09-26:
wb9vxy is mistaken.
Direwolf does NOT need hamlib for the CM108 PTT functionality.
It is builtin for Linux and Windows.
Hamlib is useful for other cases such as PTT over CAT control.