wb2osz/direwolf #284: [bug] CWID clips last character with CM108 PTT

m1geo, 2020-06-17:
When configuring DireWolf to send CWID messages using the following command in the config file:

`CBEACON delay=0:10 every=15 dest=MORSE-15 info="M1GEO"`

I observe that "M1GE" is sent and the final "O" is dropped. I suspect that this is a PTT hang time issue rather than a dropped character. I'm using a CM108 to PTT via GPIO3.

Possibly related settings:

```
ADEVICE  plughw:1,0
ARATE 48000
PTT CM108 3 /dev/hidraw0
TXDELAY 30
TXTAIL 10
```

I do not observe any clipped AX.25 packets. Just the CWID.

dranch, 2020-06-17:
What kind of computer are you running Direwolf on?  X86?  Raspberry PI older than a Pi4?  A pi4 or newer?   If you're running on an Raspberry Pi older than a Pi4, would you be willing to try out the "dead airtime" workaround (scroll down to the "OSS support" sub-section:

   http://www.trinityos.com/HAM/CentosDigitalModes/RPi/rpi4-setup.html#24a.config-direwolf

m1geo, 2020-06-17:
Hey @dranch - It's a Pi 3B+

`BCM2835 a020d3, Q1 2018, 3 Model B+, PCB 1.3, 1 GB RAM, Mfg by Sony, Wales`

Happy to try anything out if it helps the project. Is this still compatible with the CM108 PTT control?

dranch, 2020-06-17:
Yes.. still compatible.  Give it a try and let us know if that fixes your missing "O" CW character

m1geo, 2020-07-23:
Hey @dranch - I gave this a go a little while ago and got nowhere. I have just tried again, with no luck...

Here's what I did:
1. edit "Makefile.linux" to comment out "alsa = 1" to "#alsa = 1"
2. run "make tocalls-symbols"
3. run "make -f Makefile.linux -j8" _(builds cleanly)_
4. run "sudo make install" _(installs cleanly)_
5. run "sudo apt-get install alsa-oss"
6. create "/etc/asoundrc" containing "pcm.dsp0 {      type plug      slave.pcm "hw:1,0" }"
7. create "/root/.asoundrc" containing "pcm.dsp0 {      type plug      slave.pcm "hw:1,0" }" _(instructions unclear)_
8. edit /etc/direwolf.conf device line to be "ADEVICE  /dev/dsp0"
9. run "sudo reboot"
10. run "direwolf -c /etc/direwolf.conf" and receive error:
> root@aprs_pi:~# direwolf -c /etc/direwolf.conf
> Dire Wolf version 1.5
> Includes optional support for:  gpsd cm108-ptt
> 
> Reading config file /etc/direwolf.conf
> Audio device for both receive and transmit: /dev/dsp0  (channel 0)
> /dev/dsp0:
> Pointless to continue without audio device.
> root@aprs_pi:~#

11. run "aoss direwolf -c /etc/direwolf.conf" and receive error:
> root@aprs_pi:~# aoss direwolf -c /etc/direwolf.conf
> Dire Wolf version 1.5
> Includes optional support for:  gpsd cm108-ptt
> 
> Reading config file /etc/direwolf.conf
> Audio device for both receive and transmit: /dev/dsp0  (channel 0)
> Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 48000 sample rate.
> Using /dev/hidraw0 GPIO 3 for channel 0 PTT control.
> Ready to accept AGW client application 0 on port 8000 ...
> Ready to accept KISS TCP client application 0 on port 8001 ...
> 
> Now connected to IGate server london.aprs2.net (51.195.139.182)
> Check server status here http://51.195.139.182:14501
> 
> [ig] # aprsc 2.1.5-g8af3cdc
> [ig] # logresp M1GEO verified, server T2LONDON
> Can't read from audio device: Input/output error
> Terminating after audio input failure.
> root@aprs_pi:~#

Not sure what else to try at this point?
Any advice/suggestions?

George M1GEO

dranch, 2020-07-23:
Few ideas:

1. Just to confirm, you running these commands as the "root" user

2. In setting up the /etc/asoundrc file, put the matching ALSA sound device as found after running "aplay -l".  For example, "hw1,0" or "hw2,0", etc.

3. Confirm the required /dev/dsp* devices are there.  Run the command "ls /dev/dsp*"

3.a. If there isn't anything there, is the OSS kernel module installed?  Run "lsmod | grep snd_pcm_oss" and see if the module is installed.  Should look like:
--
snd_pcm_oss            45747  0
snd_mixer_oss          22468  1 snd_pcm_oss
snd_pcm                99030  6 snd_pcm_oss,snd_usb_audio,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_controller,snd_hda_codec
snd                    79517  33 snd_pcm_oss,snd_mixer_oss,snd_usb_audio,snd_usbmidi_lib,snd_rawmidi,snd_hda_codec_hdmi,snd_hda_codec_realtek,snd_hda_codec_generic,snd_hda_intel,snd_hda_codec,snd_hwdep,snd_seq,snd_seq_device,snd_pcm,snd_time
--

3.b. If it's not there (it should autoload), run the command "sudo modprobe snd_pcm_oss"

3.c. Confirm the required /dev/dsp* devices are there.  Run the command "ls /dev/dsp*"

4. Try running the "aoss direwolf -c /etc/direwolf.conf" command as root again

m1geo, 2020-07-23:
Thanks Mike. I can confirm that this fixes the issue with the morse code being clipped. Apparently the file should be /etc/asound.conf. I'll monitor the software for a few days.
George M1GEO

dranch, 2020-07-24:
Not sure who Mike is but anyway, can you elaborate what was broken so others can better understand the fix?  Also, am I reading you correctly that the required "/etc/asoundrc" file should actually be named /etc/asound.conf ?  If so, would you do me a favor and try deleting that file and see if the "/root/.asoundrc" file works instead?  If it doesn't, could you re-create the "/etc/asoundrc" file and delete the "/root/.asoundrc" file and see if things work?  This was one area that I was getting conflicting answers in my research and it would be great to set the record strait.