wb2osz/direwolf #287: Suggestion: Move Recommended Install Method to a Package Manager

aconaway1, 2020-07-22:
I've been using Direwolf as my software TNC for a while now, and it is awesome! I recommend it as often as I can.  Thumbs up to all involved!

I'm subscribed (silently) to a few mailing lists where Direwolf comes up and have noticed that there are a lot of messages asking for help in compiling from source. This is the recommended method according to the project's README.md, and I think it gets people into trouble. In general, users don't have experience in compiling code which can be quite cumbersome if you're not a software-type person. I suggest changing the README.md to recommend using a package manager (APT and YUM) to install Direwolf from repository on non-Windows and non-Mac platforms. I've used the packaged version on Raspian and Ubuntu for over a year without issue, and I think leading users to that method will help with lots of installation problems.

Keep up the good work on the project! My packets would be in a pickle without it!

73,
Aaron
KK4EHJ

dranch, 2020-07-23:
I totally understand your perspective and your ask but the answer can be complicated.  The following is from my perspective but doesn't necessarily represent Direwolf's primary developer WB2OSZ: 

Pre-built packages usually come from Linux distribution repos and not from projects themselves (aka Direwolf).  This gets additionally complicated as different distributions have different formats, release cycles, rules, etc.  Some distributions DO have Direwolf 1.5 in their repos but that version is rather old these days.   Direwolf has added lots of features and bug fixes since then but hasn't released an official 1.6 tag that many distributions require.  Even once Direwolf does release version 1.6, no currently released Linux distribution will pick it up and release it until their NEXT major release unless their package managers go to much greater lengths to get the new version "backported".  It's all rather complicated for the volunteers both on the project author's side and the various distro's packagers.  As such, many projects just "skip it" posting packages due to being too difficult.  

There are some programs out there that try to bridge this gap though.  For example,  WSJT-X.  They offer some packages for say Ubuntu 18.04, Fedora 32 and 30, and Raspbian 10/Stretch.  That's pretty impressive and helps a lot of people but there aren't packages for many other distros Ubuntu 20.04, Ubuntu 16.04, Fedora 31, Centos 8, Centos 7, Raspbian 9, Slackware, Arch, Puppy, SuSe, etc.  Overall, the community supporting WSJT-X is much larger than the community supporting Direwolf.

Maybe if various Direwolf community members are willing to submit their packaged version of Direwolf for various distros, maybe WB2OSZ would be willing to post them here.  The challenge here is that there would always be question of the integrity of those builds (are they secure, are they fully functional, etc).  Alternatively, maybe WB2OSZ could setup a daily build system using GitHub resources but that's a lot of work, probably costs a lot of money, and ultimately might not be realistic.

Just my $0.02.
--David
KI6ZHD

wb2osz, 2020-11-08:
Thanks for your feedback. I'm always interested in suggestions for improvements.  I wouldn't want your packets to be in a pickle.
Pretty much what David said.  
The README file does talk about the possibility of using apt-get or yum.   While easier, these can be very out of date, if they exist at all for your OS flavor.  
Maybe the order should be reversed.

EASY WAY:    use package install, describing the caveats. ...

PREFERRED WAY:    build from source. ...  choice of stable release or latest development for the adventurous.

Below is what I see on a Raspberry Pi.   
They haven't even caught up with version 1.5 from 2 years ago.

 $         apt-cache showpkg direwolf
Package: direwolf
Versions:
1.4+dfsg-1+b1 (/var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_buster_main_binary-armhf_Packages)
 Description Language:
                 File: /var/lib/apt/lists/raspbian.raspberrypi.org_raspbian_dists_buster_main_binary-armhf_Packages
                  MD5: 9442317f19450765ef2af3b9ac313b8c

Reverse Depends:
  hamradio-packetmodes,direwolf
  xastir-data,direwolf
  xastir,direwolf
Dependencies:
1.4+dfsg-1+b1 - libasound2 (2 1.0.16) libc6 (2 2.15) libgps23 (2 3.3) adduser (0 (null)) direwolf-docs (3 1.1-1) gpsd (0 (null)) direwolf-docs (3 1.1-1)
Provides:
1.4+dfsg-1+b1 -
Reverse Provides:

closed 2023-10-20 by aconaway1