wb2osz/direwolf #288: Usb sound card packets transmitted are heard by the soundcard

popracer, 2020-08-12:
on the usb sound card packets transmitted are heard by the sound card on raspberry pi raspbian the current release. I believe this is buster. This did not happen when I tried it on a windows installation. however I want to use the raspberry pi because it has gpio to key the radio. 

sound card is a c-media :0014 stereo sound card. 
The aplay -l shows card 1 channel 0 and the arecord -l shows card 1 channel 0 for the usb. 
so I set the config file to ADEVICE  plughw:1,0 in the first audio device properties.
It transmits and I hear the packets come over the other radio. but the software picks it up the packets immediately. almost like some kind of loop back.
My cable is set up this way: I cut a 3.5 audio patch cable in half and used both ends to connect on a circuit board the pigtail for the baofeng, and then tied the red and white wires together since it's a stereo sound card. From what I can gather it's stereo on both microphone jack and headphone jack.  on the same board there's a 2n22222 with a pull down resistor and a shunt capacitor to shunt rf.
Again this worked flawlessly on windows by setting config to use the USB sound card.

dranch, 2020-08-12:
It sounds like you're experiencing a mis-configuration of the sound device's mixer settings or some other wiring issue.  Please join and re-post your question to direwolf@groups.io as the Github "issue" area is for real software bugs or feature requests.

popracer, 2020-08-12:
Sorry thanks for your guidance.

closed 2020-08-12 by popracer

marrold, 2022-12-29:
Looking at the issues, these all seem related:

#401
#97
#288
#337

wb2osz, 2023-05-08:
Fixed - same as 401.