wb2osz/direwolf #297: FORCE_SSE is always active

ra1nb0w, 2020-11-02:
If you use this [line](https://github.com/wb2osz/direwolf/blob/413855e7915865db998239e8383cb7f84d5ac1bd/cmake/modules/FindCPUflags.cmake#L81), FORCE_SSE is always active and you can't compile direwolf with native cpu flags on i386/amd64 architecture.

The correct way is to [set](https://github.com/wb2osz/direwolf/blob/413855e7915865db998239e8383cb7f84d5ac1bd/CMakeLists.txt#L12) 
```
option(FORCE_SSE "Compile with SSE instruction only" ON)
```
then, at compile time, you can choose which flags to use or (if all are OFF) to use the auto-discovery feature.

wb2osz, 2020-11-08:
First a little history, so we can all understand the original intent before cmake was involved.
When building for an x86 32 bit target, the compiler assumed a really really ancient CPU.   I found that using the SSE instruction set made it run much faster.  (i.e. using either the -msse or -march=pentium3 gcc option)   The newer SSE instructions didn't make a difference so there is no point in using them with this application.
We have to consider the case where someone is producing a binary distribution for others.  We don't want to use -march=native because the result might not run properly on an older CPU.  We also have the case where someone is building a 32 bit distribution on a 64 bit build machine.   The generated code should not depend on the characteristics of the build machine, just the intended target.
So I think the ideal situation would be to make presence of SSE instructions or Pentium 3 the default minimum requirement for an x86 32 bit target.  Those building a distribution for others can have a minimum requirement of a Pentium 3.  Someone who has something older than a Pentium 3 should have a way to disable that minimum requirement when building for own use.  
Of course none of this applies to 64 bit target.  The generic 64 bit target implies SSE and many other instructions so nothing special would need to be done.
Am I making sense?

Maybe another way to say it would be...

For an x86 32 bit target, the default would be Pentium 3 architecture.  Building for a later 32 bit CPU does not seem to offer any benefit.  This would be good for packagers making binary distributions for others.   It would be good to have some way to turn off that requirement for someone with a computer from the previous Century.  (In all these years I only heard of one such case.)  

For an x86_64 target, just assume the generic 64 bit architecture.  That would be safe for making binary distributions for others.  I have not yet done any experiments to see if -march=native offers any benefit for a 64 bit target.

ra1nb0w, 2020-11-13:
Yes, that was the intent. But with the [line](https://github.com/wb2osz/direwolf/blob/413855e7915865db998239e8383cb7f84d5ac1bd/cmake/modules/FindCPUflags.cmake#L81) you are forcing the SSE behaviour and no options can change that fixed assignment. If I remember well, I enabled by default FORCE_SSE through option and if someone would to experiment with other settings can do that.
Therefore you should remove that line and make `option(FORCE_SSE` to ON. You obtain the same default result (if there is no other assignment elsewhere) but the user can choose which flag to use.

wb2osz, 2020-11-14:
Fixed in dev branch.   adebd06a05fd852ebdbf469ae89d5f9269661315

Thanks for all your help!

closed 2020-11-14 by wb2osz

ra1nb0w, 2020-11-14:
if you don't remove [this line](https://github.com/wb2osz/direwolf/blob/8ac14f86f5ee43176423fbdb07fe65aa18fcdb81/cmake/modules/FindCPUflags.cmake#L81) the change is useless.