wb2osz/direwolf #299: Last character of call sign dropped when using APRStt

n7nix, 2020-11-20:
Using Touch Tones for sending an APRS object using APRStt transmission the
last character of the call sign is dropped. If the call sign is padded
with a dummy character then call sign is correctly parsed. Occurs no
matter size of call sign string.

Example 1 shows problem with last character of call sign dropped.
Example 2 shows correct call sign using extra dummy character in call sign.

## 1. On sending side using call sign: N7NIX
BA236212 * A6B76B4C9B7 #
tt2text 6B76B4C9B
Decoded text from two-key method:
"N7NIX"

## 1. On receiving side
Environment variables set: TTCALL: N7NI, TTCOUNT: 1
[APRStt] KF7FIT-1>APDW16:;N7NI-12  *201725z4230.00NX13700.00WA[CN12] !TBA!

## 2. On sending side using call sign with extra character in call sign: N7NIXX
BA236212 * A6B76B4C9B9B7 #
tt2text 6B76B4C9B9B
Decoded text from two-key method:
"N7NIXX"

## 2. On receiving side
Environment variables set:: TTCALL: N7NIX, TTCOUNT: 1
[APRStt] KF7FIT-1>APDW16:;N7NIX-12 *201721z4230.00NX13700.00WA[CN12] !TBA!

I believe the problem comes from tt_user_heard() routine in file tt_user.c

Snippet from tt_user.c
/*
 * New person.  Create new table entry with all available information.
 */
strlcpy (tt_user[i].callsign, callsign, sizeof(tt_user[i].callsign));

Reference showing srlcpy copies only size - 1 characters.
strlcpy(char *dst, const char *src, size_t size);

The strlcpy() function copies up to size - 1 characters from the
NUL-terminated string src to dst, NUL-terminating the result.

Using: Dire Wolf version 1.6
Includes optional support for:  gpsd hamlib cm108-ptt

n7nix, 2020-11-21:
The following patch fixed the problem of the truncated call sign.
Tested with 4, 5 & 6 character call signs.
Using dev branch.

git diff aprs_tt.c
diff --git a/src/aprs_tt.c b/src/aprs_tt.c
index 0543f8a..bc90c7f 100644
--- a/src/aprs_tt.c
+++ b/src/aprs_tt.c
@@ -810,8 +810,8 @@ static int parse_callsign (char *e)
          }
        
          if (isupper(e[len-2])) {
-           strncpy (tttemp, e+1, len-4);
-           tttemp[len-4] = '\0';
+           strncpy (tttemp, e+1, len-2);
+           tttemp[len-2] = '\0';
            tt_two_key_to_text (tttemp, 0, m_callsign);
 
            tttemp[0] = e[len-3];

wb2osz, 2020-11-21:
The last character of the callsign is being interpreted as the symbol overlay character.
Investigating.

wb2osz, 2020-11-22:
It works as documented.  Callsign needs to be followed by a symbol overlay and checksum.
This goes way back to the original documentation here:  http://www.aprs.org/aprstt/aprstt-user.txt  and it is mentioned in the Dire Wolf APRStt implementation notes:

4.1	Traditional callsigns

The following traditional formats are recognized.   Upper case “A” means literally the “A” button.  Lower case letters are placeholders for button presses. “…” indicates it is variable length.

Att...ttvk	- Full callsign in two key method, numeric overlay, checksum.
Att...ttvvk	- Full callsign in two key method, letter overlay, checksum.

wb2osz, 2020-11-22:
A new "-d d" command line option has been added to help figure out what is going on.

----------------  Sample config file for testing ------------------------------

adevice plughw:1,0
mycall N0CAL

# Enable the APRStt conversion for channel 0.
TTOBJ  0  APP

# Recognize 6 tone grid squares.
TTMHEAD BAxxxxxx

# Simulate reception.  "t" DTI is used internally for DTMF reception.
# This works as documented.  X is the symbol overlay.
cbeacon delay=0:05 sendto=r0 info="tBA236212 * A6B76B4C9B7 #"

# Solution:  Add another character for the symbol overlay. Recompute checksum.
# Note that the symbol code is always alternate set 'A' for a box.
cbeacon delay=0:10 sendto=r0 info="tBA236212 * A 6B76B4C9B 2B 0 #"

# Alternatively, you could use the object name format and
# optionally add a more flexible symbol.
cbeacon delay=0:15 sendto=r0 info="tBA236212 * AA 6B76B4C9B * AB129 #"

-----------------------------  Run it  -----------------------------------

$ direwolf -d d -c nix.conf
Dire Wolf DEVELOPMENT version 1.7 A (Nov 21 2020)
Includes optional support for:  gpsd hamlib cm108-ptt

Reading config file nix.conf
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3, DTMF decoder enabled.
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS TCP client application 0 on port 8001 ...

[0.0] N0CAL>APDW17:tBA236212 * A6B76B4C9B7 #
Raw Touch Tone Data, DireWolf, WB2OSZ
BA236212 * A6B76B4C9B7 #
APRStt parse location (starts with B): "BA236212"
APRStt parse callsign (starts with A then digit): "A6B76B4C9B7"
Callsign in two key format1: callsign "N7NI", symbol code 'A' (Box DTMF), overlay 'X', checksum 7
[APRStt] N0CAL>APDW17:;N7NI-12  *220159z4230.00NX13700.00WA[CN12] !TBA!
[0.morse] "R"

[0.0] N0CAL>APDW17:tBA236212 * A 6B76B4C9B 2B 0 #
Raw Touch Tone Data, DireWolf, WB2OSZ
BA236212 * A 6B76B4C9B 2B 0 #
APRStt parse location (starts with B): "BA236212"
APRStt parse callsign (starts with A then digit): "A6B76B4C9B2B0"
Callsign in two key format1: callsign "N7NIX", symbol code 'A' (Box DTMF), overlay 'B', checksum 0
[APRStt] N0CAL>APDW17:;N7NIX-12 *220159z4230.00NB13700.00WA[CN12] !TBA!
[0.morse] "R"

[0.0] N0CAL>APDW17:tBA236212 * AA 6B76B4C9B * AB129 #
Raw Touch Tone Data, DireWolf, WB2OSZ
BA236212 * AA 6B76B4C9B * AB129 #
APRStt parse location (starts with B): "BA236212"
APRStt parse object name (starts with AA): "AA6B76B4C9B"
Object name in two key format: "N7NIX"
APRStt parse symbol (starts with AB): "AB129"
symbol code '=', primary symbol table '/'
[APRStt] N0CAL>APDW17:;N7NIX    *220159z4230.00N/13700.00W=[CN12] !TBA!
[0.morse] "R"

-------------------------- check results  -------------------------------------

$ echo 'N0CAL>APDW17:;N7NIX-12 *220159z4230.00NB13700.00WA[CN12] !TBA!' | decode_aprs

N0CAL>APDW17:;N7NIX-12 *220159z4230.00NB13700.00WA[CN12] !TBA!
Object, "N7NIX-12", overlayBOX DTMF & RFID & XO w/overlay B, DireWolf, WB2OSZ
N 42 30.0000, W 137 00.0000
[CN12] !TBA!

$ echo 'N0CAL>APDW17:;N7NIX    *220159z4230.00N/13700.00W=[CN12] !TBA!' | decode_aprs

N0CAL>APDW17:;N7NIX    *220159z4230.00N/13700.00W=[CN12] !TBA!
Object, "N7NIX", generic train (use steam engi, DireWolf, WB2OSZ
N 42 30.0000, W 137 00.0000
[CN12] !TBA!

n7nix, 2020-11-28:
John, thank you for the thorough explanation.
> APRStt parse callsign (starts with A then digit): "A6B76B4C9B2B0"
> Callsign in two key format1: callsign "N7NIX", symbol code 'A' (Box DTMF), overlay 'B', checksum 0

I understand that the overlay symbol follows the call sign in the two key format 1 and I am guessing that the symbol code 'A' (Box DTMF) is the default when callsign starts with an A then digit.

closed 2020-11-28 by n7nix