wb2osz/direwolf #300: cmake .. Failed on Ubuntu installation

InsertNameHere42, 2020-11-24:
setting up Direwolf on Ubuntu, and upon running CMAKE .. it gives a crash:

`CMake Error at /usr/share/cmake-3.5/Modules/ExternalProject.cmake:1888 (message):
    Could not extract tarball filename from url:

        http://www.aprs.org/aprs11/tocalls.txt
Call Stack (most recent call first):
    /usr/share/cmake-3.5/Modules/ExternalProject.cmake:2459 (_ep_add_download_command)
    data/CMakeLists.txt:41 (ExternalProject_Add)

-- Configuration incomplete, errors occurred!
See also "/home/aprs/direwolf/build/CMakeFiles/CMakeOutput.log".
See also "/home/aprs/direwolf/build/CMakeFiles/CMakeError.log".`

This is really aggravating, as I just switched from Raspberry Pi OS to Ubuntu for network support (This is on a separate PC, not a PI), and now I cannot use Direwolf without using the Ubuntu Package version which is Direwolf 1.1, Really outdated there. This fails on both Release and Dev.

dranch, 2020-11-24:
Which version of Ubuntu?  What command did you run to get this error?   Direwolf 1.6 compiles fine here on Ubuntu 20.04 using the following.  Please note that I did see a glitch on the "make update-data" stage (this is optional btw) where it was delayed on the "tocalls.txt" file as it was trying to download it from the Internet.  Could be that the upstream website might be having issues and while it worked for me, it failed for you.  

mkdir build
cd build
cmake ..
make update-data
make -j8

wb2osz, 2020-11-24:
Just skip the "make update-data" step for now.
Is it OK otherwise?

InsertNameHere42, 2020-11-24:
I didn't run the make update-data, because the cmake .. failed before I could even get to make. I believe this is Ubuntu 16.04 LTS, I'll try to update it to a newer version.

wb2osz, 2020-11-24:
What is the exact sequence of steps used?

InsertNameHere42, 2020-11-24:
```
sudo apt-get install git gcc g++ make cmake libasound2-dev libudev-dev
cd ~
git clone https://www.github.com/wb2osz/direwolf
cd direwolf
git checkout dev (Did it both with this and without, same fail each time.)
mkdir build
cd build
cmake ..
```

InsertNameHere42, 2020-11-24:
I also installed libgps-dev, gpsd, and hamlib-dev

ka1vrf, 2022-10-04:
I am having a similar issue as the previous ham. I am pasting what I am seeing: a1vrf@raspberrypi:~ $ sudo apt-get install libudev-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
libudev-dev is already the newest version (247.3-7+rpi1+deb11u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
ka1vrf@raspberrypi:~ $ cd~
bash: cd~: command not found
ka1vrf@raspberrypi:~ $ git clone https://www.github.com/wb2osz/direwolf
fatal: destination path 'direwolf' already exists and is not an empty directory.
ka1vrf@raspberrypi:~ $ cd direwolf
ka1vrf@raspberrypi:~/direwolf $ git checkout dev
fatal: Unable to create '/home/ka1vrf/direwolf/.git/index.lock': Permission denied
ka1vrf@raspberrypi:~/direwolf $ sudo git checkout dev
Branch 'dev' set up to track remote branch 'dev' from 'origin'.
Switched to a new branch 'dev'
ka1vrf@raspberrypi:~/direwolf $ sudo mkdir build && cd build
mkdir: cannot create directory ‘build’: File exists
ka1vrf@raspberrypi:~/direwolf $ cmake ..
CMake Error: The source directory "/home/ka1vrf" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
ka1vrf@raspberrypi:~/direwolf $ sudo cmake ..
CMake Error: The source directory "/home/ka1vrf" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
ka1vrf@raspberrypi:~/direwolf $ sudo make -j4
The build procedure has changed in version 1.6.
In general, it now looks like this:
 
Download the source code:
 
	cd ~
	git clone https://www.github.com/wb2osz/direwolf
	cd direwolf
 
Optional - Do this to get the latest development version
rather than the latest stable release.
 
	git checkout dev
 
Build it.  There are two new steps not used for earlier releases.
 
	mkdir build && cd build
	cmake ..
	make -j4
 
Install:
 
	sudo make install
	make install-conf
 
You will probably need to install additional applications and
libraries depending on your operating system.
More details are in the README.md file.
 
Questions?
 
 - Extensive documentation can be found in the 'doc' directory.
 - Join the discussion forum here:   https://groups.io/g/direwolf
 
ka1vrf@raspberrypi:~/direwolf $ sudo make install
make: *** No rule to make target 'install'.  Stop.
ka1vrf@raspberrypi:~/direwolf $ make install-conf
make: *** No rule to make target 'install-conf'.  Stop.
ka1vrf@raspberrypi:~/direwolf $ 

Any ideas what I did wrong?
Thanks in advance
Wayne

A-c0rN, 2022-10-04:
Not entirely sure, but when it failed to make the build directory, it seems to have NOT CDd to it. You ran CMAKE in the root directory. I would run 
```
cd ~/direwolf/build
rm -rf *
cmake ..
make -j4
sudo make install
make install-conf
```
If that doesn't work:
```
cd  ~
rm -rf direwolf
git clone https://www.github.com/wb2osz/direwolf
cd direwolf
git checkout dev
mkdir build && cd build
cmake ..
make -j4
sudo make install
make install-conf
```

wb2osz, 2023-05-09:
Older versions of cmake (< 3.10?) have a problem with the external references to  http://www.aprs.org/aprs11/tocalls.txt, etc.
These references are obsolete and have been removed in the dev branch.

Could you summarize, which versions of Ubuntu and cmake work properly, and which do not?  Thanks.

Will close this if no further discussion.