wb2osz/direwolf #303: MAX_TOCALLS needs to be larger than 200...

F4FVI, 2020-12-09:
Hello,

I have added 98 "tocalls" entries in the file tocalls.txt . (French department destination)

The problem is that now I have this message:
"MAX_TOCALLS needs to be larger than 200 to handle content of "tocalls.txt"

I opened the file "decode_aprs.c" and I modified the line  3726 from 200 to 500. 
3726 #define MAX_TOCALLS 200 -->  3726 #define MAX_TOCALLS 500

The problem is that I still have the message at the opening of direwolf "MAX_TOCALLS needs to be larger than 200 to handle content of "tocalls.txt"

Could you please help me and indicate me which paramater do I have to change to allow more entries in "tocalls.txt" file?

Thank you very much.

wb2osz, 2020-12-09:
You need to run "make" and "make install" again after changing the source code.

Be sure to save a copy of your modified /usr/local/share/direwolf/tocalls.txt  first because it will be replaced.

wb2osz, 2025-03-30:
WB4APR's tocalls.txt file has not been updated since he became a silent key.
The current authoritative source of truth is  https://github.com/aprsorg/aprs-deviceid .
This code was completely rewritten to use the new file format.
There is no arbitrary size restriction.

closed 2025-03-30 by wb2osz