wb2osz/direwolf #304: High Calibration Tone

jmkristian, 2020-12-13:
Please add an option to transmit only the high tone (2200 Hz) of 1200 bps data, for the purpose of calibration with [devcal](http://www.svxlink.org/doc/man/man1/devcal.1.html) or [by ear](https://www.febo.com/packet/layer-one/transmit.html).

On Sun, Dec 13, 2020 at 04:44 AM, Behemoth the Cat [wrote](https://groups.io/g/direwolf/message/4598):
I modified direwolf.c, so it could play both 1200Hz and 2200Hz separately in addition to what option -x does (alternating 1200/2200Hz tones - this will not work with devcal).

jmkristian, 2020-12-16:
On Tue, Dec 15, 2020 at 12:54 PM, David Ranch [wrote](https://groups.io/g/direwolf/message/4608):
Yes, summarizing all this, it would simplify a lot of things if Direwolf's calibrate feature could be enhanced to say:

   - Have one command line option for sending the MARK only tone
   - Have a different command line option for sending the SPACE only tone
   - Maybe have another command line option to offer user interaction to toggle between the MARK, SPACE, and alternating MARK/SPACE tones (like how the KPC3+ does it).

jmkristian, 2020-12-25:
Pull request #305 would resolve this issue.

wb2osz, 2021-01-01:
This is now in the dev branch.
It's a little more general because a radio channel can be specified too.
Thanks for the suggestion.

closed 2021-01-01 by wb2osz