wb2osz/direwolf #310: Audio input device 0 error code -77

nuvious, 2021-01-05:
Getting the following error on direwolf after transmitting every time I boot.

```bash
Audio input device 0 error code -77: File descriptor in bad state
Terminating after audio input failure
```
It prints the first line a bunch of times then the second line and quits. Sometimes if I reboot the raspberry pi enough it snaps out of it and returns to normal operations. My APRS setup is a raspberry pi with a USB sound card (Corsair) hooked up to a UV5R on VOX set to 1. It shows up in alsamixer and is the right card number as configured in the direwolf.conf file and I'm launching explicitly opening that file. The device is visible both before and after the failure of direwolf. Searching around I only found [this thread](https://groups.io/g/direwolf/topic/not_understanding_a_failure/34952569?p=) where someone got the same error, but I don't have kissparams or kissattach in my path, only kissutils.

For now I have a watchdog that looks for the direwolf process and if it's not running reboots my entire raspberry pi and I have startup scripts that start things in-order (with a 5 minute delay before reboot so I can debug). If the stars align it starts up properly every once in a while, runs for days, then fails and goes back into the boot loop until it gets lucky again. Would really like to resolve this issue, going to get a second usb sound card to try that in the meantime but until then any advice is appreciated!

wb2osz, 2021-01-05:
The Issues section is for reporting software bugs or making enhancement requests.  It is not a support forum.

I really should make a FAQ list and these will be the first two entries.

### My Raspberry Pi runs fine for hours/days/weeks/months and then freaks out.  What would cause this?

Most instability problems are caused by inadaquate power.  The RPi is very fussy about supply voltage,  Measure the voltage.  If you are not seeing at least 5 volts get a better power supply and a heavier wire gauge USB cable.  Try to find a power supply with 5.1 or 5.2 volts to compensate for voltage loss in the cable.  If you use a cheap crap telephone charger and ordinary charging cable you are asking for trouble, especially with the newer models that require a lot more current.  See [https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md](https://www.raspberrypi.org/documentation/hardware/raspberrypi/power/README.md)

### Direwolf receives fine on my Raspberry Pi but acts strangely after the first transmission.

This is usually caused by RF getting back into the audio interface and computer.   The transmitter might get stuck on, the audio adapter might start reporting errors, or the system might crash.  Move the radio and antenna farther away from the computer.  Try adding ferrite chokes around the cables.

wrwlfeclipse, 2021-01-16:
I'm having the same issue. Tried a powered USB Hub, tried it on a rasberry pi 3, pi zero, tried two different USB Sound cards, tried a sound card hat for raspberry pi. Verified my input power as 2.5 amp. Rewired my audio/mic plug from radio to PI and same problem. Seems like when it trys to  transmit is cause a short on the pi. Any new suggestions. Thanks.

wrwlfeclipse, 2021-01-16:
Problem Solved!!! The baofeng Mic/Headphone plug has four wires. Mic+,Mic-,Spk+,Spk-. If you have all four plugged in you will recieve the audio error. I bought a baofeng plug with no TRRS 4pole plug on end, all four wire exposed. Used Mic+,Spk+, and Mic - as Ground. Then connected to regular 3 pole jack into PI. Used the tip 1st pole on both jacks, 3rd pole for Mic- (Ground)  Send and receives with no problem. Just need to adjust volume for your mic and speaker.

nuvious, 2021-01-16:
My issue ended up getting resolved by simply buying one of those female USB A to two male USB A cables. I just plugged the data line into the pi and the other into an external USB power supply. That worked and then I got worried about the external power supply screwing with the pi's USB supply voltage and just plugged both into the pi (1 per PCB mount, not both in 1 pcb mount). Haven't seen the issue since. Definitely just a case of the USB audio card overdrawing on the USB port causing an undervoltage event. Thanks for the response and sorry for the delay in my reply!

closed 2021-01-16 by nuvious