wb2osz/direwolf #312: Version 1.6 on win10 IDP.Generic virus on direwolf.exe?

ReiniervdL, 2021-01-13:
My virus checker (Avast) quarantined direwolf.exe 1.6 on my Win10 PC.

dranch, 2021-01-13:
Can you give details of which virus Avast said the binary had?

wb2osz, 2021-01-13:
Is there some way to add a scanning exclusion for a particular directory?

Did you submit the file to Avast and tell them it was a false alarm?  This would help them improve the product.

wb2osz, 2021-02-07:
Can you give details of which virus Avast said the binary had?

Is there some way to add a scanning exclusion for a particular directory?

Did you submit the file to Avast and tell them it was a false alarm? This would help them improve the product.

This will be closed soon if no reply.

ReiniervdL, 2021-02-07:
I have provided all details that I can provide, and don’t think there’s any further action on my side. There is no way for me to check if this is a false or positive virus alert. Up to the author of direwolf to convince Avast and other virus checker providers.

73 AI6CT

wb2osz, 2021-04-10:
.

closed 2021-04-10 by wb2osz