wb2osz/direwolf #314: CPU consumption on Raspberry Pi 3 over 100% in 9600 bauds.

kovalroma, 2021-02-16:
Raspberry Pi 3
Direwolf 1.6

Command to start
`  rtl_fm -f 432.500M -g 22 -o 4 -s 44100 - | direwolf -c sdr.conf -r 44100 -B 9600 - `

After first packet received, CPU increases over 100%. Then program stucks and not working. 
On 1200 bauds working good. 
[image]

wb2osz, 2021-02-16:
I can't explain that.  This is what I get, with the same command, on an RPi 3:

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU  %MEM     TIME+ COMMAND
12674 pi        20   0  134912  12628   9424 S  43.7   1.3  61:03.90 direwolf
12673 pi        20   0   53232   6820   1824 S  11.6   0.7  15:24.04 rtl_fm

What are we doing differently?
What compiler are you using?  gcc or clang?
When cmake is run, does it say somethng about NEON?

kovalroma, 2021-02-17:
This command I use to install Direwolf
```
git clone https://www.github.com/wb2osz/direwolf
cd direwolf
mkdir build && cd build
cmake ..
make -j4
sudo make install
```
cmake log: 
```
pi@raspberrypi:~/direwolf/build $ cmake ..
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.28.0") 
-- Dire Wolf Version: 1.6.0
-- Build type set to: Release
CMake system: Linux
-- Target architecture: ARM
-- Use NEON SIMD instructions
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Checking for module 'libgps'
--   No package 'libgps' found
-- Checking for module 'hamlib'
--   No package 'hamlib' found
CMake Warning (dev) at /usr/local/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (HAMLIB)
  does not match the name of the calling package (hamlib).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/Findhamlib.cmake:55 (find_package_handle_standard_args)
  CMakeLists.txt:260 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find HAMLIB (missing: HAMLIB_LIBRARY HAMLIB_INCLUDE_DIR) 
-- Found ALSA: /usr/lib/arm-linux-gnueabihf/libasound.so (found version "1.2.3.2") 
-- Checking for module 'libudev'
--   No package 'libudev' found
CMake Warning (dev) at /usr/local/share/cmake-3.18/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (UDEV) does
  not match the name of the calling package (udev).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/Findudev.cmake:68 (find_package_handle_standard_args)
  CMakeLists.txt:274 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find UDEV (missing: UDEV_LIBRARY UDEV_INCLUDE_DIR) 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/pi/direwolf/build
```
[... 209 lines cut ...]
/home/pi/direwolf/src/symbols.c: In function ‘symbols_init’:
/home/pi/direwolf/src/symbols.c:409:6: warning: ‘strncpy’ output may be truncated copying 29 bytes from a string of length 194 [-Wstringop-truncation]
  409 |      strncpy(new_sym_ptr[new_sym_len].description, stuff+COL6_DESC, NEW_SYM_DESC_LEN);
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 94%] Building C object src/CMakeFiles/direwolf.dir/telemetry.c.o
[ 94%] Building C object src/CMakeFiles/direwolf.dir/textcolor.c.o
[ 95%] Building C object src/CMakeFiles/direwolf.dir/tt_text.c.o
[ 95%] Building C object src/CMakeFiles/direwolf.dir/tq.c.o
[ 96%] Building C object src/CMakeFiles/direwolf.dir/tt_user.c.o
[ 96%] Building C object src/CMakeFiles/direwolf.dir/xid.c.o
[ 97%] Building C object src/CMakeFiles/direwolf.dir/xmit.c.o
[ 97%] Building C object src/CMakeFiles/direwolf.dir/dwgpsnmea.c.o
[ 98%] Building C object src/CMakeFiles/direwolf.dir/dwgpsd.c.o
[ 98%] Building C object src/CMakeFiles/direwolf.dir/mheard.c.o
[100%] Building C object src/CMakeFiles/direwolf.dir/audio.c.o
[100%] Linking C executable direwolf
[100%] Built target direwolf
```

My system 
` Linux raspberrypi 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux `
```
cat /etc/issue
Raspbian GNU/Linux 8 \n \l
```