wb2osz/direwolf #315: COM PTT on Windows not working (configuration issue)

pe-jot, 2021-02-21:
First of all, I am very happy with the capabilities of Direwolf - thanks for all the efforts so far! Nevertheless, I have to file a bug or, at least, ask a question...

My setup is as following:
ICOM IC-705, connected via USB to my Windows 10 PC, IC-705 provides Audio signals and two virtual COM ports to Windows. The COM ports can be configured on the IC-705 to either have RTS or DTR acting as send (PTT) function. In my case, I've configured USB (B) RTS for this function.

Trying with my own Serial Port testing software, I was able to transmit by setting the RTS on  COM6.
However, direwolf doesn't do so, if my configuration is as following:
```
PTT COM6 RTS
```
If I start direwolf with option `-d o` I can see that my beacon is nicely surrounded by PTT 0 - 1 and PTT 0 - 0 messages, however my TRX doesn't start transmitting at all.

If I change the configuration to `PTT COM6 RTS DTR` it works fine, but I am really wondering if I have misunderstood something or if this is a configuration bug?

vy 73 de Philipp, OE5PJN

dranch, 2021-02-21:
Github issues are not for support.  They are to report real bugs or enhancement requests.  Please join the support email list at direwolf@groups.io .  That said, page 72 of the Direwolf UserGuide talks about your options including reversing the logic with the "-" option and don't forget that you might want to de-assert DTR when you assert RTS.  You need to figure out what your radio really needs.

pe-jot, 2021-02-22:
Thanks so far. Honestly, I cannot tell at the moment where the problem lies. I've done another test with an FTDI FT232R USB-Serial adapter and here it works also with `PTT COM5 RTS`. But, on the other hand, a serial terminal program which allows to manually set the RTS and DTR line is able to control the IC-705 correctly. As far as I remember it is however based on .NET Framework and it's hard to say what happens under the hood.
I'll do some debugging based on the direwolf code on my own and report the results here.

pe-jot, 2021-02-23:
Finally, I've compared the behaviour to a simple Qt C++ test application and saw that Qt handles DTR and RTS without any problems. I've cross-checked with the QtSerialPort (open)source code and updated the code of the PTT handling accordingly ;-)
It seems to me that the behaviour depends on the implementation of the serial port driver (or even the hardware) how the DTR/RTS lines behave together with Windows.

closed 2021-02-23 by pe-jot