wb2osz/direwolf #319: Alpine Build Failure

nayrnet, 2021-02-27:
Trying to make an alpine docker container, the build is failing with:

```
-- Found AVAHI: TRUE
-- Configuring done
CMake Error at external/misc/CMakeLists.txt:35 (add_library):
  No SOURCES given to target: misc
```

Here's a docker file to reproduce:
```
FROM alpine:latest as base

RUN mkdir -p /etc/direwolf
RUN apk add --no-cache  --update rtl-sdr
RUN apk add --no-cache  --update eudev
RUN apk add --no-cache  --update netcat-openbsd
RUN apk add --no-cache  --update alsa-lib
RUN apk add --no-cache  --update libusb
RUN apk add --no-cache  --update gpsd
RUN apk add --no-cache  --update avahi
#RUN apk add --no-cache  --update hamlib

FROM base as builder
RUN apk add --no-cache  --update alpine-sdk
RUN apk add --no-cache  --update cmake
RUN apk add --no-cache  --update eudev-dev
RUN apk add --no-cache  --update alsa-lib-dev
RUN apk add --no-cache  --update libusb-dev
RUN apk add --no-cache  --update gpsd-dev
RUN apk add --no-cache  --update avahi-dev
#RUN apk add --no-cache  --update hamlib-dev

RUN git clone "https://github.com/wb2osz/direwolf.git" /tmp/direwolf \
  && cd /tmp/direwolf \
  && git checkout dev \
  && cd src \
  && cd /tmp/direwolf \
  && mkdir build \
  && cd build \
  && cmake ..\
  && make -j8\
  && make DESTDIR=/target install \
  && find /target/usr/local/bin/ -type f -exec strip -p --strip-debug {} \;
```

dranch, 2021-02-27:
The misc part of the build is to add the strlcpy and strlcat functions to Linux or Windows based builds per external/misc/README .   If I was to guess, your docker setup isn't passing enough information to cmake to declare what OS you're using.  For example, here is the output of running "cmake .." on an Ubuntu 20.04 machine.  Cmake ultimately recognizes it's a Linux X86_64 setup.  What do you see on your setup?

--
$ cmake ..
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Dire Wolf Version: 1.7.0Development
-- Build type set to: Release
CMake system: Linux
-- Target architecture: x86_64
-- Use SSE SIMD instructions
-- Looking for strlcpy
-- Looking for strlcpy - not found
-- Looking for strlcat
-- Looking for strlcat - not found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Checking for module 'libgps'
--   Found libgps, version 3.20
-- Found GPSD: /usr/lib/x86_64-linux-gnu/libgps.so
-- Checking for module 'hamlib'
--   Found hamlib, version 3.3
-- Found HAMLIB: /usr/lib/x86_64-linux-gnu/libhamlib.so  
-- Found ALSA: /usr/lib/x86_64-linux-gnu/libasound.so (found version "1.2.2") 
-- Checking for module 'libudev'
--   Found libudev, version 245
-- Found UDEV: /usr/lib/x86_64-linux-gnu/libudev.so  
-- Could NOT find AVAHI (missing: AVAHI_COMMON_FOUND AVAHI_CLIENT_FOUND) 
-- Configuring done
-- Generating done
-- Build files have been written to: /usr/src/packaging/direwolf/direwolf-1.7/build
--

nayrnet, 2021-02-28:
I too see Linux x86_64, I've got a debian based container that builds perfectly fine with the same dependencies, was trying to see if I could make an alpine build for further space optimization, although I'm down to 28M w/debian I'd like to get it to build in alpine. 
```
-- The C compiler identification is GNU 10.2.1
-- The CXX compiler identification is GNU 10.2.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.30.1")
-- Dire Wolf Version: 1.7.0-04b32c4
-- Build type set to: Release
CMake system: Linux
-- Target architecture: x86_64
-- Use SSE SIMD instructions
-- Looking for strlcpy
-- Looking for strlcpy - found
-- Looking for strlcat
-- Looking for strlcat - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Checking for module 'libgps'
--   Found libgps, version 3.21
-- Found GPSD: /usr/lib/libgps.so
-- Checking for module 'hamlib'
--   Package 'hamlib', required by 'virtual:world', not found
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (HAMLIB)
  does not match the name of the calling package (hamlib).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/Findhamlib.cmake:55 (find_package_handle_standard_args)
  CMakeLists.txt:280 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Could NOT find HAMLIB (missing: HAMLIB_LIBRARY HAMLIB_INCLUDE_DIR)
-- Found ALSA: /usr/lib/libasound.so (found version "1.2.4")
-- Checking for module 'libudev'
--   Found libudev, version 243
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (UDEV) does
  not match the name of the calling package (udev).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/Findudev.cmake:68 (find_package_handle_standard_args)
  CMakeLists.txt:294 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found UDEV: /usr/lib/libudev.so
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:273 (message):
  The package name passed to `find_package_handle_standard_args` (AVAHI) does
  not match the name of the calling package (Avahi).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindAvahi.cmake:12 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:299 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found AVAHI: TRUE
CMake Error at CMakeLists.txt:326 (add_subdirectory):
  The source directory

    /tmp/direwolf/external/misc

  does not contain a CMakeLists.txt file.

-- Configuring incomplete, errors occurred!
```

wb2osz, 2023-04-10:
Is this still a problem or can this be closed?

tstellanova, 2023-06-11:
Building `dev` branch from source on debian I also saw the cmake error 
`-- Could NOT find Avahi (missing: AVAHI_COMMON_FOUND AVAHI_CLIENT_FOUND) `
and worked around this by pre-installing the dependency:
` sudo apt install libavahi-client-dev`
This would make the full dependency chain:
`sudo apt-get install gcc g++ make cmake libasound2-dev libudev-dev libavahi-client-dev` 
It may be worth updating the `README.md` instructions for `On Debian / Ubuntu / Raspbian / Raspberry Pi OS`

wb2osz, 2023-09-23:
There are multiple open cases for the same issue of building on Alpine.
These are combined into https://github.com/wb2osz/direwolf/issues/344 .
Closing this one.

closed 2023-09-23 by wb2osz