wb2osz/direwolf #323: INTERNAL ERROR? select_t1_value

jmkristian, 2021-03-02:
Direwolf logs an ominous message:
`INTERNAL ERROR?  Stream 0: select_t1_value, rc = 2, t1 remaining = -999.000, old srt = 15.000, new srt = 15.000, Extreme new t1v = 30.500`
... when attempting to connect via two digipeaters, immediately before retransmitting SABM. Excerpts from the log are in the attached file OAKND.txt.

I haven't noticed this when using one or zero digipeaters. I used Direwolf version 1.6 on Windows 8.1 Pro, with a SignaLink USB sound adapter.

Perhaps this isn't a serious issue. But it's annoying.
[OAKND.txt](https://github.com/wb2osz/direwolf/files/6065796/OAKND.txt)

dranch, 2021-03-02:
Please send the direwolf.conf file you're using.  If I was to guess, you have a misconfigured FRACK setting.

jmkristian, 2021-03-02:
Here's my direwolf.conf file. I renamed this copy because github refused to upload a .conf file.
[direwolf.conf.txt](https://github.com/wb2osz/direwolf/files/6070526/direwolf.conf.txt)

dranch, 2021-03-02:
I don't think there is a bug here and thus this shouldn't be getting discussed in a Github issue.  If you would, please close this issue, add yourself to the direwolf@groups.io email list and re-post this issue there.  

Anyway.. to conclude here, I looked at your config and it seems OK though maybe non-optimal (you seem to be using the Santa Clara County's (Bay Area, CA) overly conservative TNC settings): 
--
#SLOTTIME is in tens of milliseconds  (the setting here is 100ms) - OK
SLOTTIME 10

#PERSIST of 63 is OK
PERSIST 63

#PACLEN is in bytes at the L2 Ax.25 layer - OK
PACLEN 128

#MAXFRAME of 2 is very inefficient but OK; try using 4 on good quality RF links
MAXFRAME 2

#FRACK (aka T1) is in seconds - this setting is 6 seconds - OK
FRACK 6

#RETRY of 8 is OK
RETRY 8

#TXDELAY is in tens of milliseconds  (the setting here is 400ms) - OK
TXDELAY 40
--

The other thing to consider is what program are you using with Direwolf?  Outpost Packet Manager?  If so, Outpost might be reconfiguring the Direwolf software TNC via in-line KISS parameters.  Try this: start up direwolf in a terminal window and start capturing it's output.  Now start Outpost and attempt a packet connection.  Text will show up on the Direwolf window of what Outpost might be re-configuring.  Send that output to a post on the Direwolf email list and I bet there might be something screwy going on there.

wb2osz, 2021-03-02:
Thanks for reporting this.
I've seen this on very rare occasions.  It's a warning that an automatically adjusted parameter is outside the expected range.  After the warning is printed, the value is forced back in range.  I will put it thru a stress test and see if I can cause this to happen so it can be investigated.

jmkristian, 2021-03-02:
I use Direwolf primarily with Outpost PMM, for emergency communication in Santa Clara County. Outpost uses the AGWPE interface, not KISS. Here's a sample log:

`Dire Wolf version 1.6

Reading config file direwolf.conf
Available audio input devices for receive (*=selected):
    0: Microphone (HD Webcam C615)
    1: Digital Audio (S/PDIF) (High De
    2: Microphone (High Definition Aud
 *  3: Line (3- USB AUDIO  CODEC)   (channel 0)
    4: Line In (High Definition Audio
Available audio output devices for transmit (*=selected):
    0: Speakers (High Definition Audio
 *  1: Speakers (3- USB AUDIO  CODEC)   (channel 0)
    2: Digital Audio (S/PDIF) (High De
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate.
Note: PTT not configured for channel 0. (Ignore this if using VOX.)
Ready to accept KISS TCP client application 0 on port 8100 ...
Ready to accept AGW client application 0 on port 8000 ...

Attached to AGW client application 0 ...

Ready to accept AGW client application 1 on port 8000 ...
Attempting connect to ARCMB ...
[0L 15:33:46] W6JMK>ARCMB,LPRC2,OAKND:(SABM cmd, p=1)
INTERNAL ERROR?  Stream 0: select_t1_value, rc = 2, t1 remaining = -999.000, old srt = 15.000, new srt = 15.000, Extreme new t1v = 30.500
[0L 15:34:17] W6JMK>ARCMB,LPRC2,OAKND:(SABM cmd, p=1)
`

jmkristian, 2021-03-03:
Here's a longer, complete log, showing several connection attempts with different digipeater paths. The radio was turned off, so nothing was received. All connections were from the same client (Outpost PMM); the only difference is the digipeater paths. It seems noteworthy that only one of the three connections resulted in the ominous message.
[direwolf-log.txt](https://github.com/wb2osz/direwolf/files/6079066/direwolf-log.txt)