wb2osz/direwolf #330: An incompatibility was introduced with gpsd api 11

Thomas-McCarthy, 2021-04-12:
https://github.com/wb2osz/direwolf/blob/b66c21d39bfc0354fc582c3bc682f7963c3dda03/src/dwgpsd.c#L63

An incompatibility was introduced again when using gpsd api 11 from gpsd: 3.22.1~dev (revision release-3.22-332-g743585cc3)
Increasing the 
Changing  GPSD_API_MAJOR_VERSION > 11 to 12 solves this.

#if GPSD_API_MAJOR_VERSION < 5 || GPSD_API_MAJOR_VERSION > 12

Maybe there is a cleaner way of implementing this fix.

aryanprophet, 2021-12-21:
Changing "GPSD_API_MAJOR_VERSION > 11" to "GPSD_API_MAJOR_VERSION > 12", solved the issue for me. It compiled with a few warnings, but it works. Thank you!!!

wb2osz, 2021-12-21:
Could you be more specific about what you mean "it works?"
Do you mean that it compiles properly or did you actually run it and observe that GPS location was being put in a tracker beacon?

wb2osz, 2023-04-10:
Compatibility has been tested up through gpsd 3.25 in the dev branch.
This will be merged to master when direwolf 1.7 is released.
Closed.

closed 2023-04-10 by wb2osz