wb2osz/direwolf #332: GPS received is always same.

vipersq, 2021-04-22:
hi,
i have a small but really annoying problem, I'm using Direwolf as a receive only with SDR connected. When I transmit from my Yaesu ft3d , Direwolf always showing same coordinates. I've tried to reset FT3d to factory settings and start from beginning, I reinstall OS on raspberry and direwolf multiple times on different raspberrys , still same issue.
Gate showing in wrong place on aprs.fi as well. 
Its revieve this : N 51 00.0000, W 001 49.4100, 0 MPH, course 87, alt 331 ft , no matter what i do its always the same. 
Any ideas why?? 
M6OER.

Thomas-McCarthy, 2021-04-22:
Hi,
Take a look in your sdr.conf OR direwolf.conf to see if lat= or long= is hard coded 
The row should start with PBEACON 

Tom EI5IEB

dranch, 2021-04-22:
If you shutdown your SDR+Direwolf setup, travel around in your area, do other Igates and APRS digipeaters show your correct location?  If the answer is NO, it seems your issue is with your FT3 radio.  Please close this Github issue and if you need help with your radio, maybe someone on the direwolf@groups.io email list can give you a hand.  If the answer is YES, there could be some sort of parsing issue with your new Yaesu radio.  If that's the case, please turn on direwolf debugging with the "-d up" options and post your findings here.

wb2osz, 2021-04-22:
I understand the situation as the FT3D  transmitting, presumably with its internal GPS for the location.
Direwolf is only receiving.

https://aprs.fi/info/M6OER-7   reports:

51°04.84' N 1°49.41' W - locator IO91CB19EI - show map - static map
1.4 miles Northwest bearing 303° from Salisbury, Wiltshire, England, United Kingdom [?]
1.7 miles East bearing 87° from Wilton, Wiltshire, England, United Kingdom

Here is some raw data received by IGate

2021-04-22 07:18:38 EDT: M6OER-7>UQZZZZ,WIDE1-1,WIDE2-1,qAO,M6OER:`wMFl W[/`"4y}_0
2021-04-22 07:20:41 EDT: M6OER-7>UQZZZZ,WIDE1-1,WIDE2-1,qAO,M6OER:`wMEl L[/`"4x}_0
2021-04-22 12:05:34 EDT: M6OER-7>UQPTXT,WIDE1-1,WIDE2-1,qAO,M6OER:`wMFl v[/`"4h}_0

When we decode that, the position is changing.

N 51 00.0000, W 001 49.4200, 0 MPH, course 59, alt 322 ft
N 51 00.0000, W 001 49.4100, 0 MPH, course 48, alt 318 ft
N 51 04.8400, W 001 49.4200, 0 MPH, course 90, alt 266 ft

What am I missing?
Is the FT3D outside with a clear view of the sky?

This does not sound like a software bug so it does not belong here.
Please take it to the discussion forum   https://groups.io/g/direwolf  and be more specific about the situation.
Thanks.

vipersq, 2021-04-23:
thx all for answering, for starter i can't check if other stations are receiving me as there is no APRS stations in Salisbury UK , :(.
But I found an issue , not with direwolf but with FT3d, i had to factory reset and update firmware, now it seems to be working fine. 
THX for help.

closed 2021-04-23 by wb2osz