wb2osz/direwolf #334: TBEACON sending invalid packets

mytechguyri, 2021-05-06:
I've had a working install of v1.6, and recently added a GPS.  I changed one of my PBEACON entries to a TBEACON to test the GPS functionality, and aprs.fi receives invalid packets.

Here is the packet recieved by APRS.fi with PBEACON.. everything as it should be.
2021-05-06 02:42:05 CDT: WA1OKB-10>APDW16,TCPIP*,qAC,T2USANE:!S9GH\<IjY#{6G/A=000003WA1OKB Oakland Beach-Warwick, RI - Pi Zero Digi/iGate https://www.wa1okb.radio/home/aprs

and here's what happens when I switch to TBEACON.... invalid packets
2021-05-06 02:19:10 CDT: WA1OKB-10>APDW16,TCPIP*,qAC,T2SJC:!S!!!!!!!!#{6G/A=000000WA1OKB Oakland Beach-Warwick, RI - Pi Zero Digi/iGate https://www.wa1okb.radio/home/aprs [Latitude range check failed]

The only change in the config between TBEACON and PBEACON is that I've obviously removed the lat and long values since it would get those from the GPS, everything else is exactly the same between the valid PBEACON and the invalid TBEACON

My logs show Direwolf reports a 3D fix with gpsd

GPS appears to be working as expected... here is the output from cgps... I'm stumped as to what the problem is.
┌───────────────────────────────────────────┐┌──────────────────Seen 16/Used  5┐
│ Time:        2021-05-06T14:05:31.000Z (18)││GNSS   PRN  Elev   Azim   SNR Use│
│ Latitude:         41.68639418 N           ││GP  7    7  47.0  309.0  27.0  Y │
│ Longitude:        71.39971241 W           ││GP  8    8  67.0  189.0  16.0  Y │
│ Alt (HAE, MSL):    -72.254,     32.182 ft ││GP  9    9  42.0  243.0  27.0  Y │
│ Speed:             1.97 mph               ││GP 16   16  38.0   62.0  16.0  Y │
│ Track (true, var):   311.4, -14.2     deg ││GP 27   27  67.0   69.0  25.0  Y │
│ Climb:            -1.18 ft/min            ││GP  4    4  30.0  201.0   0.0  N │
│ Status:         3D FIX (5 secs)           ││GP 18   18   6.0   33.0   0.0  N │
│ Long Err  (XDOP, EPX):  0.96, +/- 47.5 ft ││GP 21   21   9.0  162.0   0.0  N │
│ Lat Err   (YDOP, EPY):  1.54, +/- 75.7 ft ││GP 26   26  17.0   77.0   8.0  N │
│ Alt Err   (VDOP, EPV):  3.78, +/-  206 ft ││GP 30   30  10.0  308.0   0.0  N │
│ 2D Err    (HDOP, CEP):  1.82, +/-  173 ft ││SB133   46  35.0  217.0   0.0  N │
│ 3D Err    (PDOP, SEP):  4.20, +/-  243 ft ││SB135   48  12.0  250.0   0.0  N │
│ Time Err  (TDOP):       2.99              ││SB138   51  30.0  227.0   0.0  N │
│ Geo Err   (GDOP):       5.15              ││QZ  2  194   n/a    0.0   0.0  N │
│ ECEF X, VX:    1521543.750 m   -0.030 m/s ││QZ  4  196   n/a    0.0   0.0  N │
│ ECEF Y, VY:   -4521068.720 m    0.070 m/s ││QZ  5  197   n/a    0.0   0.0  N │
│ ECEF Z, VZ:    4219634.990 m   -0.020 m/s ││                                 │
│ Speed Err (EPS):       +/-  103 mph       ││                                 │
│ Track Err (EPD):       +/-  107 deg       ││                                 │
│ Time offset:            0.132547452 s     ││                                 │
│ Grid Square:            FN41hq24          ││                                 │
└───────────────────────────────────────────┘└─────────────────────────────────┘
{"class":"SKY","device":"/dev/ttyACM0","time":"2021-05-06T14:07:47.000Z","xdop":0.93,"ydop":1.05,"vdop":2.67,"tdop":2.01,"hdop":1.42,"gdop":3.62,"pdop":3.02,"nSat":16,"uSat":6,"satellites":[{"PRN":4,"el":29.0,"az":201.0,"ss":19.0,"used":true,"gnssid":0,"svid":4,"health":1},{"PRN":7,"el":48.0,"az":309.0,"ss":28.0,"used":true,"gnssid":0,"svid":7,"health":1},{"PRN":8,"el":69.0,"az":189.0,"ss":21.0,"used":true,"gnssid":0,"svid":8,"health":1},{"PRN":9,"el":41.0,"az":242.0,"ss":26.0,"used":true,"gnssid":0,"svid":9,"health":1},{"PRN":16,"el":37.0,"az":63.0,"ss":10.0,"used":true,"gnssid":0,"svid":16,"health":1},{"PRN":18,"el":6.0,"az":32.0,"ss":0.0,"used":false,"gnssid":0,"svid":18,"health":1},{"PRN":21,"el":10.0,"az":162.0,"ss":0.0,"used":false,"gnssid":0,"svid":21,"health":1},{"PRN":26,"el":16.0,"az":78.0,"ss":8.0,"used":false,"gnssid":0,"svid":26,"health":1},{"PRN":27,"el":67.0,"az":67.0,"ss":22.0,"used":true,"gnssid":0,"svid":27,"health":1},{"PRN":30,"el":10.0,"az":308.0,"ss":16.0,"used":false,"gnssid":0,"svid":30,"health":1},{"PRN":46,"el":35.0,"az":217.0,"ss":0.0,"used":false,"gnssid":1,"svid":133,"health":1},{"PRN":48,"el":12.0,"az":250.0,"ss":0.0,"used":false,"gnssid":1,"svid":135,"health":1},{"PRN":51,"el":30.0,"az":227.0,"ss":0.0,"used":false,"gnssid":1,"svid":138,"health":1},{"PRN":193,"az":0.0,"ss":0.0,"used":false,"gnssid":5,"svid":1,"health":1},{"PRN":194,"az":0.0,"ss":0.0,"used":false,"gnssid":5,"svid":2,"health":1},{"PRN":197,"az":0.0,"ss":0.0,"used":false,"gnssid":5,"svid":5,"health":1}]}
{"class":"TPV","device":"/dev/ttyACM0","mode":3,"time":"2021-05-06T14:07:47.000Z","leapseconds":18,"ept":0.005,"lat":41.686394182,"lon":-71.399712412,"altHAE":-22.0229,"altMSL":9.8091,"alt":9.8091,"epx":13.994,"epy":15.825,"epv":63.084,"track":311.3699,"magtrack":297.1819,"magvar":-14.2,"speed":0.879,"climb":-0.006,"epd":48.0568,"eps":31.65,"epc":126.17,"ecefx":1521543.44,"ecefy":-4521049.43,"ecefz":4219623.26,"ecefvx":-0.07,"ecefvy":-0.11,"ecefvz":-0.21,"ecefpAcc":25.91,"ecefvAcc":0.90,"velN":0.581,"velE":-0.660,"velD":0.006,"geoidSep":-31.832,"eph":52.988,"sep":74.320}

mytechguyri, 2021-05-06:
I did do some further checking, since the output from cgps did not look correct to me, I set my GPS to send NMEA packets...  so now the output looks more like I would expect
(68) $GPGSV,3,1,11,01,05,171,,04,15,195,12,07,63,304,24,08,85,150,21*77
(66) $GPGSV,3,2,11,09,30,227,19,14,03,272,10,16,26,074,,21,24,155,*79
(55) $GPGSV,3,3,11,26,06,087,,27,53,051,15,30,23,310,26*43
(36) $GPGST,144133.00,55,,,,10,15,17*7A
(38) $GPZDA,144133.00,06,05,2021,00,00*64
(40) $GPGBS,144133.00,10.4,14.8,16.7,,,,*79
(68) $GPRMC,144134.00,A,4141.17761,N,07123.97341,W,1.291,,060521,,,A*67
(35) $GPVTG,,T,,M,1.291,N,2.390,K,A*20
(75) $GPGGA,144134.00,4141.17761,N,07123.97341,W,1,06,1.47,11.0,M,-33.7,M,,*5E
(54) $GPGSA,A,3,30,07,09,27,04,08,,,,,,,2.46,1.47,1.97*0B

but still the same result.... here is from the direwolf log... The packet at 10:37:16 is a PBEACON the packet at 10:37:44 is the failing TBEACON

May  6 10:37:16 direwolf direwolf[2769]: [0L] WA1OKB-10>APDW16,WIDE1-1,WIDE2-1:!S9GH\<IjY#{6G/A=000003Oakland Beach Warwick, RI - Pi Zero  Digi/iGate https://www.wa1okb.radio/home/aprs
May  6 10:37:17 direwolf direwolf[2769]: GPSD: Location fix is now 3D.
May  6 10:37:19 direwolf direwolf[2769]: Now connected to IGate server noam.aprs2.net (199.204.38.35)
May  6 10:37:19 direwolf direwolf[2769]: Check server status here http://199.204.38.35:14501
May  6 10:37:22 direwolf direwolf[2769]: [rx>ig] user WA1OKB-10 pass 24238 vers Dire-Wolf 1.6 filter t/omsn/WA1OKB-10/50
May  6 10:37:23 direwolf direwolf[2769]: [ig] # aprsc 2.1.5-g8e78d39
May  6 10:37:23 direwolf direwolf[2769]: [ig] # logresp WA1OKB-10 verified, server T2OREGON
May  6 10:37:44 direwolf direwolf[2769]: [ig] WA1OKB-10>APDW16:!S!!!!!!!!#{6G/A=000000WA1OKB Oakland Beach-Warwick, RI - Pi Zero Digi/iGate https://www.wa1okb.radio/home/aprs

dranch, 2021-05-06:
If you look at your original CGPS output, the grid square was putting you in Argentina and not Rhode Island!  Next, can you show use what your PBEACON and TBEACON lines in your direwolf.conf file are?

Regardless, this seems to be fixed as everything looks ok here:
   https://aprs.fi/?c=raw&call=WA1OKB-10&limit=1000&view=normal

If you agree, please close this Github issue.  If you have more questions, please take it to direwolf@groups.io as this is most likely NOT a direwolf bug.

mytechguyri, 2021-05-06:
> please take it to ***@***.*** as this is most likely NOT a

Thank you David... No it's not fixed, I switched back to PBEACON, that's
why it looks ok now.

Here is the PBEACON line that works

PBEACON delay=0:01 every=10:00 overlay=S symbol=\# power=13 lat=41.68618
long=-71.39961 alt=1 height=25 gain=6 compress=1 comment="Oakland Beach
Warwick, RI - Pi Zero iGate"

Here is the TBEACON line that does not.
TBEACON delay=0:01 every=10:00 overlay=S symbol=\# power=13 alt=1 height=25
gain=6 compress=1 comment="Oakland Beach Warwick, RI - Pi Zero iGate"

Literally the only difference is T instead of P and remove lat and long.

dranch, 2021-05-06:
Please review the Direwolf User Guide section 9.9.4 but the syntax for TBEACON is somewhat simplified than the PBEACON or OBEACON syntax.  If I was to guess, your issue stems from your comment being too long for the APRS frame.  Try removing it from your configuration and see if TBEACON starts working.  If it does, start adding a very short comment and work your way up until Direwolf breaks.

wb2osz, 2021-05-06:
pi@raspberrypi:~ $ echo 'WA1OKB-10>APDW16,WIDE1-1,WIDE2-1:!S9GH<IjY#{6G/A=000003' | decode_aprs

WA1OKB-10>APDW16,WIDE1-1,WIDE2-1:!S9GH<IjY#{6G/A=000003
Position, Fog (& future ovrly codes) w/overlay S, DireWolf, WB2OSZ
N 41 41.1759, W 018 32.3071, 20 MPH, course 84
A=000003

Latitude is good but the longitude is off.
This location is in the ocean, west of Spain.

I've never seen anyone else use the compressed format so it could be that there has been a bug all these years and no one ever noticed.

As a workaround, take out the compress=1 to get the human readable format.  This needs to be investigated.

mytechguyri, 2021-05-07:
Okay, I'll give that a try.
I pared down the comment to just my call sign, so TBEACON line now reads

TBEACON sendto=IG delay=0:30 every=60:00 symbol=\# overlay=S alt=1 power=13
height=25 gain=6 compress=1 comment="WA1OKB gps"

Same problem

So then I removed compress

TBEACON sendto=IG delay=0:30 every=60:00 symbol=\# overlay=S alt=1 power=13
height=25 gain=6 comment="WA1OKB gps"

For some reason it's not pulling the correct data from gpsd

May  6 20:57:31 direwolf direwolf[617]: [rx>ig]
WA1OKB-10>APDW16,qAR,WA1OKB-10:!00  nanNS000  nanE#360/000/A=000000WA1OKB
gps

mytechguyri, 2021-05-07:
Even though it does say GPSD has a fix

May 6 21:05:03 direwolf direwolf[2331]: GPSD: Location fix is now 3D.

wb2osz, 2021-05-07:
We have a problem here that needs to be fixed.

Question:
**Did you build this from source or install a package (e.g. yum, apt-get) from somewhere?**

The GPSD people keep making incompatible changes to the API.
If you look for occurrences of  GPSD_API_MAJOR_VERSION in /src/dwgpsd.c you will see all the changes necessary to keep up.
I'm wondering if there might be a mismatch between gpsd and libgpsd or between libgpsd installed and the version used to build direwolf.

We can find the header file API version like this:

pi@raspberrypi:~ $ grep -H GPSD_API_MAJOR_VERSION /usr/lib/*gps*  /usr/local/lib/*gps*
grep: /usr/lib/*gps*: No such file or directory
pi@raspberrypi:~ $ grep -H GPSD_API_MAJOR_VERSION

We can find the library files like this but I don't know how to determine the API version.

pi@raspberrypi:~ $ ls -l /usr/lib/libgps*  /usr/local/lib/libgps*
ls: cannot access '/usr/lib/libgps*': No such file or directory
lrwxrwxrwx 1 root root     23 Oct 27  2020  /usr/local/lib/libgpsdpacket.so -> libgpsdpacket.so.27.0.0
lrwxrwxrwx 1 root root     23 Oct 27  2020  /usr/local/lib/libgpsdpacket.so.27 -> libgpsdpacket.so.27.0.0
-rwxr-xr-x 1 root root  42412 Oct 27  2020  /usr/local/lib/libgpsdpacket.so.27.0.0
lrwxrwxrwx 1 root root     16 Oct 27  2020  /usr/local/lib/libgps.so -> libgps.so.27.0.0
lrwxrwxrwx 1 root root     16 Jun  6  2020  /usr/local/lib/libgps.so.25 -> libgps.so.25.0.0
-rwxr-xr-x 1 root root 105184 Jun  6  2020  /usr/local/lib/libgps.so.25.0.0
lrwxrwxrwx 1 root root     16 Oct 27  2020  /usr/local/lib/libgps.so.27 -> libgps.so.27.0.0
-rwxr-xr-x 1 root root 105204 Oct 27  2020  /usr/local/lib/libgps.so.27.0.0

wb2osz, 2021-05-07:
Adding "-dgt"to the direwolf command line might provide some useful clues.

mytechguyri, 2021-05-07:
So I did not build from source... The Raspbian repository has DW 1.6, and gpsd, so I installed from apt.
I am running Raspbian Bullseye, so perhaps that is the problem..... I hadn't really thought of that..... different version in the Bullseye testing release perhaps.
According to that, I've got libgps-dev and gpsd version 3.22-3

I'll burn a new image with Buster and see if that was the problem....

wb2osz, 2021-05-07:
Just build direwolf from source that that should take care of any version mismatch.
When I have a chance, I need to see if there is a way to query the libgps API version and detect if it is different than the version used to compile direwolf.

christopherbach, 2021-06-26:
I had the same issue. Compiled latest version of GPSD on a Raspberry Pi. Compiled Direwolf. Direwolf did not compile due to version mismatch of a GPSD library. 

I followed the workaround mentioned here: https://github.com/wb2osz/direwolf/pull/331?_pjax=%23js-repo-pjax-container  which is changing GPSD_API_MAJOR_VERSION from "11" to "12" in the dwgpsd.c file.

Afterwards Direwolf compiles fine and gets the correct position again.