wb2osz/direwolf #336: 9600 Bd with IC-9100 not possible?

oe1rsa, 2021-05-17:
I am trying to work 9600Bd packet radio with an ic-9100. This rig has a built in usb sound card. I was successful at 1200 Bd and hamlib for PTT control.

For 9600 Bd, however, it just doen't work at all. Same antenna, same location, same packet node, but kenwood rig works flawlessly. So my question is: Am I doing something wrong?

What I have observed: The Tx signal from direwolf to rig looks rectangular, letting me believe direwolf has no built in raised cosine filter (or any other ISI avoiding filter).

wb2osz, 2021-05-17:
The Issues section is for reporting software bugs and making enhancement requests.   This type of question should really go to the discussion forum.

9600 bps can be frustrating because it requires a very wide flat audio bandpass out to about 5 kHz.
FM transceivers distort the audio signal with preemphasis and deemphasis which make it unusable for 9600 bps.
If there is no way to disable this, for the built in USB audio, you will need to use the 6 pin mini-DIN "DATA2" connector.

According to the IC-9100 instruction manual, page 167, you also need to enable 9600 bps operation.
What version of direwolf are you using?  Anything recent should generate the signal using raised cosine shaping.

closed 2021-05-17 by wb2osz

dranch, 2021-05-17:
You will get better support by posting your support request to direwolf@groups.io than you will here as Github issues are only for bug reports.