wb2osz/direwolf #342: KISSPORT does not override 8001

dranch, 2021-07-07:
Per an email from Roger, N1XP from direwolf@groups.io, configuring an alternative KISSPORT only adds the new TCP port but the default port of 8001 isn't disabled.  I have confirmed this only occurs on v1.7-A:

--
$ sudo direwolf -t 0 -d o -a 100 -c /etc/ax25/direwolf.conf
Dire Wolf DEVELOPMENT version 1.7 A (Jul  7 2021)
Includes optional support for:  gpsd cm108-ptt

Reading config file /etc/ax25/direwolf.conf
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
Ready to accept AGW client application 0 on port 8003 ...
Ready to accept KISS TCP client application 0 on port 8001 ...  <---------------------------------
Ready to accept KISS TCP client application 0 on port 8004 ...
--

Here is a confirmation of the open ports:
--
$ sudo lsof -nPi | grep direwolf
direwolf  792   root    3u  IPv4  18033      0t0  TCP *:8003 (LISTEN)
direwolf  792   root    7u  IPv4  20010      0t0  TCP *:8001 (LISTEN)  <----------------
direwolf  792   root    9u  IPv4  18035      0t0  TCP *:8004 (LISTEN) 
--

This seems like this is a bug introduced in Direwolf v1.7 as I don't see this behavior with Direwolf v1.6:
--
$ sudo direwolf -t 0 -d o -a 100 -c /etc/ax25/direwolf.conf
Dire Wolf version 1.6
Includes optional support for:  gpsd cm108-ptt

Reading config file /etc/ax25/direwolf.conf
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, E+, 44100 sample rate / 3.
Ready to accept AGW client application 0 on port 8003 ...
Ready to accept KISS TCP client application 0 on port 8004 ...  <--------------------
--

jschollenberger, 2021-11-16:
Still an issue on version 1.7 B (Oct 26 2021). I actually came here to report what I though was a different bug in that it throws this confusing error message:

```Bind failed with error: 10048
Some other application is probably already using port 8001.
Try using a different port number with KISSPORT in the configuration file.
Ready to accept AGW client application 0 on port 8002 ...
Ready to accept KISS TCP client application 0 on port 8003 ...
```

It seems you've discovered the real issue... that it really is trying to bind to 8001. I confirmed that by only running one instance with KISSPORT set to 8003 and I see it bind to both.

j4nr6n, 2023-02-08:
Still seems to be an issue.
 `Dire Wolf DEVELOPMENT version 1.7 E (Feb  7 2023)`

j4nr6n, 2023-02-08:
See [this groups.io thread](https://groups.io/g/direwolf/topic/alternate_kiss_port/90387650) that I found in #443 
> However there is an easy workaround to prevent DireWolf from opening the default KISS port 8001:
```
# Do not open the default KISS port 8001
KISSPORT 0 
# Open alternate KISS port 8100
KISSPORT 8100
```

mfncooper, 2025-07-28:
This seems to be a documentation issue rather than a code issue. The "workaround" referred to by Thomas Leibold in Groups.io and cited above by @j4nr6n is apparently the intended design. Ever since the capability for multiple KISS ports was introduced, the code has included the comment:

    // "KISSPORT 0" is used to remove the default entry.

See https://github.com/wb2osz/direwolf/blob/dev/src/config.c#L4956