wb2osz/direwolf #344: Make Failure On alpine linux Arm64

DevenBL, 2021-08-08:
This is the output of cmake .. and make consecutively
[compiletime.txt](https://github.com/wb2osz/direwolf/files/6950171/compiletime.txt)
this is the master branch following these instructions:
git clone https://github.com/wb2osz/direwolf.git/
cd direwolf
mkdir build && cd build
cmake ..
make -j4
sudo make install
make install-conf

I have also tried the dev Branch but it fails on the cmake

DevenBL, 2021-08-08:
this is the Direct output of make for convenience :
/home/devuno/Documents/direwolf/src/direwolf.h:302:56: error: expected declaration specifiers or '...' before string constant
  302 | #define strlcat(dst,src,siz) strlcat_debug(dst,src,siz,__FILE__,__func__,__LINE__)
      |                                                        ^~~~~~~~
compilation terminated due to -fmax-errors=1.
make[2]: *** [external/misc/CMakeFiles/misc.dir/build.make:76: external/misc/CMakeFiles/misc.dir/strlcpy.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/home/devuno/Documents/direwolf/src/direwolf.h:302:56: error: expected declaration specifiers or '...' before string constant
  302 | #define strlcat(dst,src,siz) strlcat_debug(dst,src,siz,__FILE__,__func__,__LINE__)
      |                                                        ^~~~~~~~
compilation terminated due to -fmax-errors=1.
make[2]: *** [external/misc/CMakeFiles/misc.dir/build.make:90: external/misc/CMakeFiles/misc.dir/strlcat.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:491: external/misc/CMakeFiles/misc.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  2%] Building C object external/geotranz/CMakeFiles/geotranz.dir/polarst.c.o
[  3%] Building C object external/geotranz/CMakeFiles/geotranz.dir/tranmerc.c.o
[  3%] Building C object external/geotranz/CMakeFiles/geotranz.dir/ups.c.o
[  4%] Building C object external/geotranz/CMakeFiles/geotranz.dir/usng.c.o
[  4%] Building C object external/geotranz/CMakeFiles/geotranz.dir/utm.c.o
[  5%] Linking C static library libgeotranz.a
[  5%] Built target geotranz
make: *** [Makefile:156: all] Error 2

dranch, 2021-08-08:
What compiler and version are you using on your version of AlpineLinux?   You also might try compiling Direwolf v1.7 from the Git "develop" branch to see if that helps.

DevenBL, 2021-08-08:
i am using gcc 10.3.1 20210625 (Alpine 10.3.1_git20210625)

there is no "develop" branch although there is a dev branch and that fails
at the cmake with a cmake error regarding: No SOURCES given to target: misc

DevenBL, 2021-08-08:
I have also tried to compile the direwolf 1.5 and it fails in the same way as the master branch

DevenBL, 2021-08-08:
this seems to be the same issue as **https://github.com/wb2osz/direwolf/issues/150** 
when i try to cmake on the dev branch this is the output: 
-- The C compiler identification is GNU 10.3.1
-- The CXX compiler identification is GNU 10.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found Git: /usr/bin/git (found version "2.32.0") 
-- Dire Wolf Version: 1.7.0-b66c21d
-- Build type set to: Release
CMake system: Linux
-- Target architecture: ARM64
-- Use NEON SIMD instructions
-- Looking for strlcpy
-- Looking for strlcpy - found
-- Looking for strlcat
-- Looking for strlcat - found
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Checking for module 'libgps'
--   Found libgps, version 3.22
-- Found GPSD: /usr/lib/libgps.so
-- Checking for module 'hamlib'
--   Found hamlib, version 4.2
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (HAMLIB)
  does not match the name of the calling package (hamlib).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/Findhamlib.cmake:55 (find_package_handle_standard_args)
  CMakeLists.txt:284 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found HAMLIB: /usr/lib/libhamlib.so  
-- Found ALSA: /usr/lib/libasound.so (found version "1.2.5.1") 
-- Checking for module 'libudev'
--   Found libudev, version 243
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (UDEV) does
  not match the name of the calling package (udev).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/Findudev.cmake:68 (find_package_handle_standard_args)
  CMakeLists.txt:298 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found UDEV: /usr/lib/libudev.so  
CMake Warning (dev) at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (AVAHI) does
  not match the name of the calling package (Avahi).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindAvahi.cmake:12 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  CMakeLists.txt:303 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found AVAHI: TRUE  
-- Configuring done
CMake Error at external/misc/CMakeLists.txt:35 (add_library):
  No SOURCES given to target: misc

CMake Generate step failed.  Build files cannot be regenerated correctly.

DevenBL, 2021-08-08:
this is the package list: 
abuild
alpine-base
alsa-lib
alsa-lib-dev
avahi
avahi-dev
binutils
build-base
cmake
eudev
eudev-dev
eudev-libs
eudev-openrc
extra-cmake-modules
gcc
gpsd
gpsd-dev
hamlib
hamlib-dev

dranch, 2021-08-08:
Seems there are other reports of this issue with Direwolf and it seems to be more of a Cmake on Alpine issue:

   https://www.gitmemory.com/issue/wb2osz/direwolf/319/787401401

If you look at the internals of Direwolf's external/misc/CMakeLists.txt file, you see explicit mention of this issue with Alpine Linux starting at line 16:
--
# It seems that Alpine Linux and Void Linux have strlcpy and
# strlcat so we need to handle the situation more delicately.
# When doing it this way, there is probably no reason to
# distinguish between Linux and BSD-like systems here.
# If we kept going, the same thing could be done for each
# of the functions and no OS check would be needed.
--

There are hints in https://github.com/flann-lib/flann/issues/369 saying that the issue is with the way Cmake is being used here but it's not 100% clear what the proper fix for Direwolf would be.

dranch, 2021-08-08:
Seems like some hints on the fix can be found here but I cannot try a fix as I don't have Alpine LInux:

https://bugs.gentoo.org/show_bug.cgi?id=652594 is the overview and the specific example fixes for that project are here:

   https://github.com/alpinelinux/aports/pull/3901/files

DevenBL, 2021-08-08:
> Seems like some hints on the fix can be found here but I cannot try a fix as I don't have Alpine LInux:
> 
> https://bugs.gentoo.org/show_bug.cgi?id=652594 is the overview and the specific example fixes for that project are here:
> 
> https://github.com/alpinelinux/aports/pull/3901/files

Alpine linux uses cmake-3.21.1-r0

this seems to explain the issue succinctly https://github.com/buildroot/buildroot/commit/0c469478f64d0ddaf72c0622a1830d855306d51c 
it mentions "But flann CMake use a trick that use
an empty string "" as source list"
and that's probably what's at the core of this issue: https://github.com/flann-lib/flann/issues/369

TLDR: Cmake changed the way add_library() works.

"With CMake >= 3.11, the new behaviour of add_library() break the
existing flann CMake code." - https://github.com/buildroot/buildroot/commit/0c469478f64d0ddaf72c0622a1830d855306d51c

DevenBL, 2021-08-08:
so as per the error "CMake Error at external/misc/CMakeLists.txt:35 (add_library):
No SOURCES given to target: misc" seems like the misc target needs a source

DevenBL, 2021-08-08:
Ok so i just tested it on Archlinux and it doesn't seem to be related to the cmake version since it compiles fine on arch might have to do Busybox being different to the Gnu Core Utils

DevenBL, 2021-08-08:
going back to fixing the compile issue on the master branch and not the cmake issue on the Dev branch
this seems relevant:
https://github.com/void-linux/void-packages/pull/27510

DevenBL, 2021-08-08:
This is the same issue aswell https://github.com/wb2osz/direwolf/issues/319

DevenBL, 2021-08-08:
[output.txt](https://github.com/wb2osz/direwolf/files/6951617/output.txt)
For the sake of completeness i am slamming all of the output from the master branch in here this is the output of 
cmake ..
make -j4
make -j4
Here is the output from the last make: 
[  1%] Building C object external/misc/CMakeFiles/misc.dir/strlcat.c.o
[  1%] Building C object external/misc/CMakeFiles/misc.dir/strlcpy.c.o
[  5%] Built target geotranz
/home/devuno/Documents/direwolf/src/direwolf.h:302:56: error: expected declaration specifiers or '...' before string constant
  302 | #define strlcat(dst,src,siz) strlcat_debug(dst,src,siz,__FILE__,__func__,__LINE__)
      |                                                        ^~~~~~~~
/home/devuno/Documents/direwolf/src/direwolf.h:302:56: error: expected declaration specifiers or '...' before string constant
  302 | #define strlcat(dst,src,siz) strlcat_debug(dst,src,siz,__FILE__,__func__,__LINE__)
      |                                                        ^~~~~~~~
compilation terminated due to -fmax-errors=1.
compilation terminated due to -fmax-errors=1.
make[2]: *** [external/misc/CMakeFiles/misc.dir/build.make:76: external/misc/CMakeFiles/misc.dir/strlcpy.c.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [external/misc/CMakeFiles/misc.dir/build.make:90: external/misc/CMakeFiles/misc.dir/strlcat.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:491: external/misc/CMakeFiles/misc.dir/all] Error 2
make: *** [Makefile:156: all] Error 2

makerio90, 2023-08-13:
arch. both aur options fail and manual install also fails.

wb2osz, 2023-09-23:
I installed Alpine Linux in a VM so I can try to get to the bottom of this.
Forgive me if I;m going back to the beginning covering the same territory again.  I'm trying to understand what is going on here,
I'm using the "dev" branch which will be release 1.7.  

cmake says error at external/misc/CMakeLists.txt (add_library)
No sources given to target: misc

The problem is that we are adding an empty list to add_library.
This is cmake version 3.26.5.  We see some discussion above, and other Internet rumblings, about the change in functionality when the source file list is empty.

We have a similar situation with Mac OSX where the C library provides strlcpy.  In my case cmake version is 3.17.0.

AHA!  When we look in external/misc/CMakeLists.txt, we see there is a test for OS type.  For BSD and similar, we simply assume the strl... functions are available, and have:

set(MISC_LIBRARIES "" CACHE INTERNAL "")

rather than following the Linux logic.  Perhaps the solution would be to do this instead of add_library if the source file list is empty.

73,
john WB2OSZ

wb2osz, 2023-09-23:
It should also be noted that these packages need to be installed:
linux-headers
alsa-lib-dev

This was added to the 1.7 User Guide (out soon).

wb2osz, 2023-09-23:
I pushed a change (commit 877d1c7707ac11df9d32b821e3c6dfd66e561e09) that fixed the build for Alpine Linux and I thought that should be ok for other platforms.
Unfortunately it broke Ubuntu and Raspberry Pi.
We need a cmake expert to take a look at this.

wb2osz, 2023-09-25:
I think I have something that fixes the build on Alpine without breaking the others.
Try the most recent in the "dev" branch.