wb2osz/direwolf #345: Improve cm108 utility's flexibility for testing PTT, etc

dranch, 2021-08-10:
Would it be difficult to add a feature to the cm108 tool to try setting the GPIO pin to test PTT?  

Maybe add some options like:
    -v : version of this tool
    -d : the specific device to control
    -g : the GPIO pin to control (default to GPIO3)
    -a : action: 1 for high; 0 for low

I'm thinking it would be cool if it could maybe have two modes to help users confirm they have the right HID device for PTT control:

   - Specific device using the -d option : using the spacebar, toggle the state of the GPIO pin.  Use control-c to return GPIO pin back to it's original state and then exit

   - Interactive Probe (if -d isn't specified):
      - Cycle through each GPIO pin on each found highdraw device that offers GPIO control
       Using the spacebar, each of the GPIO pin's state will be toggled until the spacebar is pressed again.  At that point, the GPIO pin will be put back to it's previous state and it will toggle the state of the next GPIO pin.  Once all available GPIO pins are touched, move on to the next valid hidraw interface.  Use control-c to return GPIO pin back to it's original state and then exit

tylerhunt, 2023-01-03:
For what it's worth, you can use `rigctl` from [Hamlib](https://hamlib.github.io) to test PTT.

The command `T` sets the PTT. Use `1` for on and `0` for off:

```
$> rigctl -m 6 -P CM108 -p /dev/hidraw0

Rig command: T
PTT: 1

Rig command: T
PTT: 0
```

According to the documentation, you should be able to use `-C ptt_bitnum=3` to change the GPIO pin. This is a 0-indexed value. The default is `2` which is GPIO 3 on pin 13. This functionality [appears to be broken](https://github.com/Hamlib/Hamlib/issues/1203), however.

dranch, 2023-01-03:
I think this is an interesting alternative but I don't think the current CM108 support is via Hamlib and instead, it's via direct USB HID access.  Is that correct John?

wb2osz, 2023-01-04:
direwolf and the cm108 utility access the Cmedia chip through the HID interface.   They don't hamlib to do that.

dranch, 2023-01-04:
I'm not sure if the posted Windows binary for 1.7A has Hamlib support but if it does and if there is indeed a cm108 issue with the Windows binary, maybe this could be an alternative approach to help those users

wb2osz, 2023-01-04:
The Windows version does not support hamlib. 
I downloaded  the first zip file  from  https://github.com/wb2osz/direwolf/releases and extracted the files.
With a DINAH interface connected, I ran the cm108 utility:

```
C:\Users\John\Downloads\direwolf-1.7.0-dev-A_i686\direwolf-1.7.0-7fa91dd_i686>cm108
    VID  PID   Product                          HID [ptt]
    ---  ---   -------                          ---------
    0461 4d15  USB Optical Mouse                \\?\hid#vid_0461&pid_4d15#8&14905c6b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
    413c 2010  Dell USB Keyboard                \\?\hid#vid_413c&pid_2010&mi_00#9&2c9e8850&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\kbd
    413c 2010  Dell USB Keyboard                \\?\hid#vid_413c&pid_2010&mi_01&col01#9&14dafc0b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
    413c 2010  Dell USB Keyboard                \\?\hid#vid_413c&pid_2010&mi_01&col02#9&14dafc0b&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}
    413c 2010  Dell USB Keyboard                \\?\hid#vid_413c&pid_2010&mi_01&col03#9&14dafc0b&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}
**  0d8c 0012  USB Audio Device                 \\?\hid#vid_0d8c&pid_0012&mi_03#8&2a5a85b6&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}
```

Next, I ran cm108 again, this time adding the USB Audio Device HID address to the command line:

```
C:\Users\John\Downloads\direwolf-1.7.0-dev-A_i686\direwolf-1.7.0-7fa91dd_i686>cm108  "\\?\hid#vid_0d8c&pid_0012&mi_03#8&2a5a85b6&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}"
0101010101010101010101010
```

Notice the following:
(1) The HID path needs to be surrounded by quotation marks (") because it contains some characters with special meaning to the command interpreter.
(2) Alternating 0 and 1 are displayed as the PTT control is turned off and on.  

The DINAH red LED goes on and off.