wb2osz/direwolf #348: Windows Build Freezes Randomly (1.7)

na7q, 2021-08-28:
I'm using the latest beta release for Windows. It seems that it will randomly freeze up and become completely unresponsive after X amount of minutes or hours. Typically it freezes in less than 24 hours, with little use, or no use. I'm using it as a simple RMS gateway TNC for Winlink. I'm using it on 2 machines, and both are different machines but are identical in problems.
I'm using Windows 10.

dranch, 2021-08-28:
Please see https://github.com/wb2osz/direwolf/issues/260#issuecomment-907558318 which I bet is a similar issue to yours.   If you agree, can you close this issue and then follow / comment to your specific scenario there?  If that other issue DOESN'T seem to cover your issue, you can keep this issue open but Github "issues" are not for asking for help.  They are intended for reporting real bugs or requesting new features.  Instead, consider joining the direwolf@groups.io email list and ask for help there.

na7q, 2021-08-29:
I'm unsure if they are related at this time. As I have had one Windows computer that doesn't transmit, receive only for testing, and it has frozen also. I'll continue to monitor. If I can't reproduce it when no RF is nearby I will close it.

na7q, 2021-09-07:
Can't reproduce yet. So I will close it until it I can reproduce it.

closed 2021-09-07 by na7q