wb2osz/direwolf #349: Windows Sound Card Specification Undocumented or Unsupported

na7q, 2021-08-30:
Seems there is no way to specify the sound cards for PTT on Windows. It automatically and always selects the opposite one. 
My config:

ADEVICE USB PnP Sound Device USB PnP Sound Device
ACHANNELS 1
CHANNEL 0
PTT CM108

I've tried every combination I can think of to make this work. No success. There is no documentation of this for Windows. So I assume it's just not supported.

[image]

na7q, 2021-08-31:
To add more details. I did open the CM108 utility and tested PTT with the HID path and it keys fine there. Using the HID path in the Direwolf config results in nothing but an error, expecting a /dev/hidraw path.

dranch, 2021-08-31:
This is also being discussed at https://groups.io/g/direwolf/topic/85250767 in greater detail.  Need to hear from WB2OSZ on next steps

na7q, 2021-09-01:
Best temporary workaround I have found is to use a different USB port. As long as the PTT HID path I want to use is LAST in the list in Direwolf, then it will use that for PTT.