wb2osz/direwolf #351: Direwolf 1.7.0-dev-A on Windows still decoding with no input audio feed ?

zanco, 2021-09-02:
Hi, every now and then I fly and track APRS High Altitude Balloons. Sometimes I connect my Direwolf to my virtual audio card and decode APRS from websdr receivers to check if my tracker shows up there.

Today I had Firefox running, playing audio to virtual audio and Direwolf was set to take the input from the same virtual audio device, Works like a charm. After some time I decided to stop decoding so I closed Firefox. The audio on my speakers from the websdr disappeared, the speakers are playing the same virtual audio so that makes sense.

However, now, 35 minutes after closing the browser, the Direwolf window still shows messages ?

> [ig>tx] G3CWI-5>APMI06,TCPIP*,qAS,G3CWI:@020749z5315.80N/00208.79W&Macclesfield iGate. Supply 12.3 Volts.

[ig>tx] G4WLC-1>APMI06,TCPIP*,qAC,T2POLAND:@020751z5145.38N/00231.08W_306/001g003t054r000p000P000h88b10297Oldcroft WX

[ig>tx] M0GQS>APMI06,TCPIP*,qAC,T2LONDON:@020752z5126.27N/00057.83W-WX3in1Plus2.0 U=14.1V, T=33.8C

[ig>tx] G3CWI-5>APMI06,TCPIP*,qAS,G3CWI:@020754z5315.80N/00208.79W&Macclesfield iGate. Supply 12.3 Volts.

Can anyone explain this ? Is there a large buffer somehow ?

Thanks,

Ben - PE2BZ

dranch, 2021-09-02:
The github "issues" are not for requesting support.  Please post any help reqests to direwolf@groups.io.  Regardless, you issue you're reporting is not an issue.. you have enabled Igate connectivity through the Internet and the packets you see of "IG>TX" is means APRS packets from the Internet are intended to be transmitted over RF by your station.  If this isn't desired, you should disable the APRS-IS interface within the direwolf.conf file.  Please close this ticket.

zanco, 2021-09-02:
Typical noob user error. Srry