wb2osz/direwolf #352: Error preparing after bad state: File descriptor in bad state

kuhnje, 2021-09-07:
Sound dongle is good, power is good (using a powered USB hub, but even without power is good and getting same error).
Dire Wolf DEVELOPMENT version 1.7 A (Sep  7 2021)
-------
(on a RPI3 with a USB sound dongle and EasyDigi with USB PTT):
-------
pi@RPi3:~ $ dmesg | grep C-Media
[    3.820421] usb 1-1.1.2: Manufacturer: C-Media Electronics Inc.
[    3.826205] input: C-Media Electronics Inc. USB Audio Device as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.1/1-1.1.2/1-1.1.2:1.3/0003:0D8C:0014.0001/input/input0
....
--------
(It is seen):
---------
pi@RPi3:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
.
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
--------
(and):
----------
pi@RPi3:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
----------
(telling direwolf about it in direwolf.conf)
------------
ADEVICE  plughw:1,0
ACHANNELS 1
CHANNEL 0
MODEM 1200
PTT /dev/ttyUSB0 RTS DTR
------------
(Runs fine and APRS program connects fine)
------------
Reading config file direwolf.conf
Audio device for both receive and transmit: plughw:1,0  (channel 0)
Channel 0: 1200 baud, AFSK 1200 & 2200 Hz, A+, 44100 sample rate / 3.
Ready to accept AGW client application 0 on port 8000 ...
Ready to accept KISS TCP client application 0 on port 8001 ...
-------------
(but after telling it to transmit a beacon):
-------------
Audio input device 0 error code -19: No such device
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Terminating after audio input failure.
-------------
Audio dongle works fine when I test it with speaker-test using pair of headphones.  Program connects fine to it on port 8001.  PTT works correctly.  
Its has to be something in direwolf.  I tried using either:
ADEVICE  plughw:1,0
(or)
ADEVICE - plughw:1,0
Since there are references to using both, but using either ends in:
-------------
Error preparing after bad state: File descriptor in bad state
Audio write error retry count exceeded.
-------------

dranch, 2021-09-07:
This looks like a classic RFI issue when your radio transmits, it's impacting your USB sound device.  Please review the many of other discussions on the direwolf@groups.io email list for possible mitigations, etc.  Please close this Github issue as these are only used for real Direwolf bugs or enhancement requests.

kuhnje, 2021-09-07:
> This looks like a classic RFI issue when your radio transmits, it's impacting your USB sound device.  Please review the many of other discussions on the direwolf@groups.io email list for possible [...]

Ok, thanks.  I'll take it to the email list.

closed 2021-09-07 by kuhnje