wb2osz/direwolf #358: Display CRC bytes?

K4KDR, 2021-11-11:
Good day!

Is there any way to actually DISPLAY the CRC bytes that were transmitted by a sender?

The question is not whether a packet's CRC is correct - but rather a tool is needed here to actually VIEW the CRC bytes so that we can confirm which CRC calculation is being included with outgoing packets.

Many thanks!

-Scott,  K4KDR

wb2osz, 2021-11-11:
Look in hdlc_rec.c.    Somewhere around line 562, depending on the exact version you have.
Here the actual received and expected FCS values are compared.

	    actual_fcs = H->frame_buf[H->frame_len-2] | (H->frame_buf[H->frame_len-1] << 8);
	    expected_fcs = fcs_calc (H->frame_buf, H->frame_len - 2);

	    if (actual_fcs == expected_fcs) {
	      alevel_t alevel = demod_get_audio_level (chan, subchan);
	      multi_modem_process_rec_frame (chan, subchan, slice, H->frame_buf, H->frame_len - 2, alevel, RETRY_NONE, 0);   /* len-2 to remove FCS. */
	    }
	    else {
#if TEST
	      dw_printf ("*** actual fcs = %04x, expected fcs = %04x ***\n", actual_fcs, expected_fcs);
#endif
	    }

Delete the two lines beginning with "#" then rebuild.
If the FCS is not correct, the actual and expected values will be printed.
You could also move the dw_printf before the "if" to print the values all the time.

closed 2021-11-11 by wb2osz

K4KDR, 2021-11-11:
WONDERFUL!!

Thank you so much!

K4KDR, 2021-11-11:
Thank you again for pointing me to the file where the 'fcs' bytes can be printed.

I have tried positioning that print statement in numerous places above & inside of those 'if' statements, but nothing results in the expected printout.  This is a fresh 'git clone' of the latest MASTER branch.

One example of my edit is:

[image]

... but after deleting the /build folder and completely rebuilding/reinstalling, the output is not changed:

[image]

... did I do that incorrectly?

Thanks so much!

-Scott

K4KDR, 2021-11-11:
Found a work-around!

If I understand correctly, that entire section of code was controlled by a flag called "OLD_WAY", which was not set.

I commented that out, along with a couple of corresponding lines farther down, and then found a place that will print the CRC bytes with each valid frame (don't know why they print 9 times, but that's ok)

[image]

... and my output:

[image]

... so that will do the job until / unless you have time one day to provide a more proper method.  This is extremely useful for examining packets in regard to what is 'really' being transmitted over-the-air.  

Thanks very much!

-Scott