wb2osz/direwolf #360: Audio output data underrun in Linux

minorsecond, 2021-11-19:
When using either Direwolf 1.5 or the latest master in Linux, an audio output data underrun occurs when transmitting. This does not occur in Windows

- OS: Arch Linux
- Rig: FT-818
- Direwolf versions: 1.5, master
- Alsa audio

LongnoseRob, 2021-11-19:
Hello,

could you provide output from direwolf, ideally with the `-a 10`option set?
Playback of other sounds/music via alsa makes no problems?

minorsecond, 2021-11-19:
Yes, other audio sources play fine.
I've attached the direwolf output with the `-a 10` flag set.

[direwolf_log.txt](https://github.com/wb2osz/direwolf/files/7570087/direwolf_log.txt)

LongnoseRob, 2021-11-19:
which sound-hardware are you using?
Have you tried a different sound card?

minorsecond, 2021-11-19:
It's the XGGCOMS Digimode-4 interface.

I don't have another. An interesting note is that Direwolf was working with this soundcard in Fedora.

LongnoseRob, 2021-11-19:
have you checked for any messages regarding this interface in dmesg?
have you updated the packages lately?
would you be able to try the dev-branch built from source?

minorsecond, 2021-11-19:
Nothing from dmesg other than when the device is attached on boot. I'll build the dev branch and test.

minorsecond, 2021-11-19:
On the latest dev I get the same issue.

minorsecond, 2021-11-19:
I just tested on the LTS kernel (5.10), and it's working. I'm wondering if they updated the kernel drivers for my audio device between 5.10 and 5.15 and broke it.

LongnoseRob, 2021-11-19:
@minorsecond Good, atleat know you have a workaround..

minorsecond, 2021-11-19:
In the meantime, I'm filing a kernel bug report. Thanks for your help!

closed 2021-11-19 by minorsecond

minorsecond, 2021-11-19:
The bugzilla report is here, if anyone comes across this issue and wants to follow:

https://bugzilla.kernel.org/show_bug.cgi?id=215073

dranch, 2021-11-20:
There are still some missing details here that would be good to understand:
   - Which version of Arch Linux is this?  Which specific kernel version were you seeing this issue with?
   - What hardware are you running this on?  X86?  ARM?  If say a Raspberry Pi 3+ or older, you might consider the alternative USB stack to see if that helps

A few more things:  It's highly recommended to run Direwolf v1.6 as it has many fixes over v1.5.  Next, your kernel bug report is incorrect as you are experiencing sound device under runs but your report gave specifics around the FTDI serial device.   http://www.xggcomms.com/userimages/USB-Digimode-4.pdf says this device uses the CM108 sound chipset.  Finally, you might consider putting some ferrites over each cable end of this sound device to potentially avoid any RFI issues.

minorsecond, 2021-11-20:
Hi dranch,

Arch is rolling release, so there isn't a specific version. Everything matches upstream pretty closely as far as software versions go. The kernel version that doesn't work is 5.15.2. I can reproduce this easily by switching kernels on boot. It works with the 5.10, but not 5.15.

The hardware is an Intel NUC, so it shouldn't be underpowered.

Thanks for the info about the bug report. I'll update it.

minorsecond, 2021-11-30:
The issue appears to be fixed in kernels 5.15 and 5.16 by adding the following to /etc/modprobe.d/usb-audio.conf and then rebooting:

`echo "options snd-usb-audio lowlatency=n" /etc/modprobe.d/usb-audio.conf`

ds2k5, 2025-03-05:
could this the same issue in Kernel Version 6.8.0  too ?