wb2osz/direwolf #361: autostart

surucu, 2021-11-22:
Hi
I ran direwolf aprs version 1.7, very successful, but I can't restart with crontab and dw-start.sh with restart. It starts manually, I wonder where am I doing wrong. Is there a solution to this?

wb2osz, 2021-11-22:
The "issues" section is for reporting software defects and enhancement requests.  This is not a support forum.  

Questions like this should go to the discussion forum,  https://groups.io/g/direwolf

There are many people who can help.  You can search for problems and solutions.

First troubleshooting step:   EXACTLY what is in crontab?  Type "contrab -l" and copy/paste the result.

dranch, 2021-11-29:
Also, when there are errors in a cronjob, there should be errors reported in /var/log/syslog.  You can quickly scan those results by running:

   sudo cat /var/log/syslog | grep CRON

If you find anything, please post that in a new thread on the direwolf@groups.io email list.