wb2osz/direwolf #364: Minimum packet length configurable?

TomasOK2PNQ, 2021-12-10:
We are using Direwolf to decode downlink from satellites, for example GRBAlpha. But it is using CSP protocol, where minimum length of the packet is just 8 bytes including CRC. The Direwolf does not decode so short packets, minimum lenght we observed is 15 bytes including CRC. Would it be possible to make the length configurable?
Thanks a lot in advance
              Tomas OK2PNQ

wb2osz, 2021-12-10:
The minimum valid length for a frame, excluding the CRC, is controlled by

#define AX25_MIN_PACKET_LEN ( 2 * 7 + 1 )

in ax25_pad.h.   This is based 2 addresses of 7 bytes each and a control byte.
A quick workaround would be to change that to 6 and rebuild.

If that works, we can talk about adding a configuration option.

dranch, 2021-12-10:
Do you have a URL for the specification of this CSP protocol?

TomasOK2PNQ, 2021-12-13:
Hi wb2osz and dranch,
I asked others, which are more informed about the CSP things, I got this as response, I hope it will help you...

A rather good start is the wikpiedia, as always https://en.wikipedia.org/wiki/Cubesat_Space_Protocol

Basically, the CSP packets are nothing but raw HDLC packets like in the case of AX.25. So, in direwolf, the CRC-16-CCITT is also part of the CSP in this sense (since it confirms to another layer) and direwolf indeed checks it.

The structure is basically this:

HDLC packet == HDLC payload + CRC-16-CCITT

HDLC payload == CSP header + CSP payload + CSP CRC32

or

HDLC payload == AX.25 packet

(see all of our SatNOGS decoded packets, those are basically HDLC packets, regardless whether AX.25 or CSP).

CSP header == 4 bytes (https://en.wikipedia.org/wiki/Cubesat_Space_Protocol#Protocol_header)

CSP payload == arbitrary length (from 0 up to anything)

CSP CRC32 == CRC-32C (Castagnoli), w/ normal representation of 0x1EDC6F41, 4 bytes

wb2osz, 2021-12-13:
I was looking through some of the documentation, including external links, and it was not clear what was relevant.
If you simply need HDLC with CRC-16-CCITT, direwolf would probably be applicable with trivial modification.
If CRC-32 is required, that could turn out to be lot of effort because the 16 bit CRC is referenced so many places.
Do you have audio recordings of some sample signals?
Did you try changing the minimum frame length (AX25_MIN_PACKET_LEN) to a smaller value?  If this works, I could add a configuration option to make this a runtime variable instead.

wb2osz, 2023-05-19:
If you simply need HDLC with CRC-16-CCITT, direwolf would probably be applicable with trivial modification.
If CRC-32 is required, that could turn out to be lot of effort because the 16 bit CRC is referenced so many places.
Do you have audio recordings of some sample signals?
Did you try changing the minimum frame length (AX25_MIN_PACKET_LEN) to a smaller value? If this works, I could add a configuration option to make this a runtime variable instead.