wb2osz/direwolf #371: PTT via HamLib configuration - TTY with - in name

darksidelemm, 2022-01-06:
My IC-9700 shows up as `/dev/tty.usbserial-5320`.

I've tried the following configuration lines:
```
PTT RIG 3081 "/dev/tty.usbserial-5320"
PTT RIG 3081 /dev/tty.usbserial-5320
```

In both cases, direwolf interprets the serial port as being `/dev/tty.usbserial-`, resulting in:
```
Hamlib Rig open error -6: IO error
port_open: serial_open(/dev/tty.usbserial-) status=-6, err=No such file or directory
iofunc.c(81):port_open return(-6)
rig.c(800):rig_open return(-6)
```

I'm unsure if this is an issue in Hamlib, or an issue in direwolf's configuration parsing. Yes I know that this can be worked around with a symlink, which is what I'm doing at the moment, but that device name is valid, so should work...

wb2osz, 2022-01-07:
What version of direwolf are you using?

darksidelemm, 2022-01-07:
`Dire Wolf DEVELOPMENT version 1.4 E (Jan  2 2022)`
(Installed via Homebrew on OSX)

wb2osz, 2022-01-07:
That's the problem.
There was a bug where the ptt device name would be truncated at 19 characters.
According to the comments, it was fixed in version 1.5 (released October 2018).

Answer is to upgrade to 1.5 or newer if you want to stop using your work-around.

https://github.com/wb2osz/direwolf/releases

closed 2022-01-07 by wb2osz

darksidelemm, 2022-01-07:
Ahh that would explain it... I saw the 2nd Jan 2022 and thought I was on a recent version... nope. Sorry for the annoyance!