wb2osz/direwolf #372: Audio output will abruptly stop on occasion when running Direwolf on a Raspberry Pi 4 or laptop

ElectricCarbon, 2022-01-15:
I am using Direwolf for sending audio through a USB soundcard to my radio. I have tried using the VOX on the radio and the Hamlib library to enable CAT push to talk in the `direwolf.conf` file. Around 75% of the time the audio output from direwolf will work with no problems. However, occasionally the audio will cut out either at the beginning of the carrier tone or halfway through sending a packet. I have checked to see if this is a issue with resource consumption but the CPU and memory are only at 15% when Direwolf is transmitting. When using hamlib the push to talk on my radio will stay on if the audio cuts out.

I have the same issue on my laptop however the audio ending abruptly occurs less frequently then on the Pi. I have the same configuration file for my laptop and the Pi. I have also tried using two different USB soundcards including my laptop's built-in speaker output and the issue is still present.

I am running direwolf using:
`direwolf -B 1200`

dranch, 2022-01-17:
Github "issues" is not how you get support for Direwolf.  Please close this issue, subscribe to direwolf@groups.io and re-post your question there.  When you do, please cite what radio you're using and how you've cabled things between your computer and your radio.

wb2osz, 2022-01-17:
Please provide exact details on how to set up a situation where this occurs.
Example:  Try this

cbeacon delay=0:10 every=0:03 info=" 1 The quick brown fox jumps over the lazy dog "

Is it consistent or are some packets short?
Whenever there is inconsistent transmitting behavior we have to rule out RF getting back into the computer.
How far is the antenna from the radio?
Does the problem persist if the transmitter is changed to low power?