wb2osz/direwolf #374: Digipeater ambiguity: "Digipeater WIDE# (probably XXXXX-Y)"

Tyler-2, 2022-01-27:
```
Digipeater WIDE2 (probably K2VIZ-8) audio level = 93(44/24) [NONE] __||||||_
[0.4] W4RAT-2>APOT30,K2VIZ-8,WIDE2*:!3751.64N/07732.43W#W2 RATS.NET Beaverdam VA
Position, DIGI (white center), Open Track
N 37 51.6400, W 077 32.4300
W2 RATS.NET Beaverdam VA
```

and

```
Digipeater WIDE1 (probably K2VIZ-8) audio level = 94(41/25) [NONE] __||||||_
[0.4] K4EME-3>BEACON,K2VIZ-8,WIDE1*,WIDE2-1:!3809.92N/07918.85W#PHG5850/WIDE-RELAY digi on Elliott Knob,VA A=4440<0x0d>
```

```
Digipeater K4EME-3 audio level = 85(36/21) [NONE] ____||___
[0.4] KV3B-2>APN383,K4EME-3*,WIDE2:!3857.05NS07652.41W#PHG5560 W2, MDn-N, MARC Digi East MD<0x0d>
Position, OVERLAY DIGI (green star) w/overlay S, Kantronics KPC-3 rom versions, 25 W height=320 6dBi omni
N 38 57.0500, W 076 52.4100
W2, MDn-N, MARC Digi East MD
```

What is it about these first two examples that causes the ambiguity about who they're from? I think Direwolf is correct about the "probably" but is there something improper/invalid that leads to this ambiguity? When fed into Xastir, Xastir doesn't recognize these as not-digipeated packets, which is the only reason I noticed them.

The owner of that station says "Great questions. The APRS standard isn't absolute and has some gray areas between manufacturers. K2VIZ-8 is running a TNC-X, and as the "hop" is used, WIDE2-1 would be converted to WIDE2."
but that doesn't sound strange, so doesn't explain why this ambiguity exists.

wb2osz, 2022-01-27:
This really does not belong here because it is not reporting a defect or making an enhancement request.
It really belongs in the discussion forum.  

APRS digipeating is not well defined so there are different interpretations.
Look in Dire Wolf User Guide, sections:
9.5.8.     Digipeater algorithm
9.5.9      APRS Digipeater - Compared to other implementations

The information that someone is using TNC-X is not relevant.  It is not capable of digipeating on its own.  It needs to be done by some application.

Tyler-2, 2022-01-28:
Ok, if the "bug" here is in the inherent ambiguity of APRS, I can accept that it's not a problem for Direwolf to fix. I don't understand the behavior well enough. Thanks!

closed 2022-01-28 by Tyler-2