wb2osz/direwolf #392: "Audio input device 0 error: No such device" after PTT button was is triggered

diablowu, 2022-05-12:
Description: 
When I trigger PTT (manually pressing the PTT key or automatically triggering VOX through PBEACON), direwolf will have a sound card device error：
```
Audio input device 0 error code -19: No such device
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Audio input device 0 error code -77: File descriptor in bad state
Terminating after audio input failure.
```

And then the process  will exit.

/var/log/message:

```
May 12 08:56:54 raspberrypi kernel: [ 6142.215047] retire_capture_urb: 10 callbacks suppressed
May 12 08:56:54 raspberrypi kernel: [ 6142.238137] Transfer to device 9 endpoint 0x6 frame 248 failed - FIQ reported NYET. Data may have been lost.
May 12 08:56:54 raspberrypi kernel: [ 6142.239136] Transfer to device 9 endpoint 0x6 frame 249 failed - FIQ reported NYET. Data may have been lost.
May 12 08:56:54 raspberrypi kernel: [ 6142.240146] Transfer to device 9 endpoint 0x6 frame 250 failed - FIQ reported NYET. Data may have been lost.
May 12 08:56:54 raspberrypi kernel: [ 6142.241136] Transfer to device 9 endpoint 0x6 frame 251 failed - FIQ reported NYET. Data may have been lost.
May 12 08:56:54 raspberrypi kernel: [ 6142.242136] Transfer to device 9 endpoint 0x6 frame 252 failed - FIQ reported NYET. Data may have been lost.
May 12 08:56:54 raspberrypi kernel: [ 6142.243145] Transfer to device 9 endpoint 0x6 frame 253 failed - FIQ reported NYET. Data may have been lost.
May 12 08:56:54 raspberrypi kernel: [ 6142.355739] usb 1-1.3: reset full-speed USB device number 9 using dwc_otg
May 12 08:56:55 raspberrypi kernel: [ 6142.895648] usb 1-1.3: reset full-speed USB device number 9 using dwc_otg
May 12 08:56:56 raspberrypi kernel: [ 6143.435657] usb 1-1.3: reset full-speed USB device number 9 using dwc_otg
May 12 08:56:56 raspberrypi kernel: [ 6143.975679] usb 1-1.3: reset full-speed USB device number 9 using dwc_otg
May 12 08:56:57 raspberrypi kernel: [ 6144.416451] usb 1-1.3: USB disconnect, device number 9
May 12 08:56:57 raspberrypi kernel: [ 6144.535786] usb 1-1.3: new full-speed USB device number 10 using dwc_otg
May 12 08:56:57 raspberrypi kernel: [ 6145.085713] usb 1-1.3: new full-speed USB device number 11 using dwc_otg
May 12 08:56:58 raspberrypi kernel: [ 6145.526029] usb 1-1-port3: attempt power cycle
May 12 08:56:58 raspberrypi kernel: [ 6146.185752] usb 1-1.3: new full-speed USB device number 12 using dwc_otg
May 12 08:56:59 raspberrypi kernel: [ 6146.924388] usb 1-1.3: New USB device found, idVendor=1b3f, idProduct=2008, bcdDevice= 1.00
May 12 08:56:59 raspberrypi kernel: [ 6146.924454] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
May 12 08:56:59 raspberrypi kernel: [ 6146.924484] usb 1-1.3: Product: USB Audio Device
May 12 08:56:59 raspberrypi kernel: [ 6146.924510] usb 1-1.3: Manufacturer: GeneralPlus
May 12 08:56:59 raspberrypi kernel: [ 6146.988991] input: GeneralPlus USB Audio Device as /devices/platform/soc/20980000.usb/usb1/1-1/1-1.3/1-1.3:1.3/0003:1B3F:2008.0009/input/input27
May 12 08:56:59 raspberrypi kernel: [ 6147.056885] hid-generic 0003:1B3F:2008.0009: input,hidraw4: USB HID v2.01 Device [GeneralPlus USB Audio Device] on usb-20980000.usb-1.3/input3
May 12 08:56:59 raspberrypi mtp-probe: checking bus 1, device 12: "/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3"
May 12 08:56:59 raspberrypi mtp-probe: bus: 1, device: 12 was not an MTP device
May 12 08:57:00 raspberrypi mtp-probe: checking bus 1, device 12: "/sys/devices/platform/soc/20980000.usb/usb1/1-1/1-1.3"
May 12 08:57:00 raspberrypi mtp-probe: bus: 1, device: 12 was not an MTP device
```

OS: Raspbian OS 32bit (Linux raspberrypi 5.15.32+ #1538 Thu Mar 31 19:37:58 BST 2022 armv6l GNU/Linux)
Hardware: Raspberry Pi Model B Rev 2
External USB sound card: 1b3f:2008 Generalplus Technology Inc. USB Audio Device
RIG: BaoFeng UV5R
Direwolf:  Tag 1.6  build on above Raspberry PI 
config:
```
ADEVICE plughw:2,0
ACHANNELS 1
CHANNEL 0
MYCALL BI1NIZ-3
MODEM 1200
```

dranch, 2022-05-12:
Github issues are not a place to get support but a place to report real bugs or request enhancement requests.  To get future support, please join the Direwolf email list direwolf@groups.io.  Anyway, the issue you are reporting smacks of RFI getting into your sound device and/or Raspberry Pi's USB bus and making the the sound device reset.  Try getting the radio as far away from your Raspberry Pi as possible, try adding some ferrites on the all cables coming IN/OUT of the Raspberry Pi, try lowering the TX power of the radio, etc.  If you need more help, please re-post this on the email list.  Please close this Github issue.

wb2osz, 2022-05-12:
As David said, it is most likely RF getting into the USB and causing an error.  Please take it to the discussion forum.

closed 2022-05-12 by wb2osz