wb2osz/direwolf #395: How to accomplish APRS proportional pathing?

Tyler-2, 2022-05-21:
As mentioned here: http://www.aprs.org/newN/ProportionalPathing.txt

It is a good idea for some objects to be beaconed with much shorter paths more frequently, and only going wider paths with less frequency. This allows, for instance, a traveler heading into an area, to have a list of digis in the area they're heading towards, while also ensuring that someone who has just turned on their radio will get `local` info.

Configuring this in Direwolf doesn't seem directly possible.... I did something like this:
```
#OBEACON sendto=0 compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" slot=0:10 via=DIRECT every=1
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" slot=0:15 via=WIDE1-1 every=3
OBEACON sendto=0 compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" slot=0:20 via=WIDE1-1,WIDE2-1 every=9
OBEACON sendto=0 compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" slot=0:25 via=WIDE1-1,WIDE2-2 every=27
```

This is a little ugly and tough to manage changes in, and Direwolf complains that there need to be a whole number worth of beacons every hour, and then, if you have a few of these, when the `WIDE2-2` fires, the `DIRECT` and `WIDE1-1` and `WIDE2-1` beacons are also firing.

Obviously if every minute we want to go `DIRECT`, and every 3 minutes we want to go `WIDE1-1`, I don't want to fire beacons for `DIRECT` path _and_ `WIDE1-1` path at minute 3...

How can this be worked around?

wb2osz, 2022-05-24:
The User Guide has an example of how to do proportional pathing.

PBEACON  DELAY=1  EVERY=30  VIA=WIDE1-1 LAT=42^37.14N  LONG=71^20.83W  
PBEACON  DELAY=11  EVERY=30                         LAT=42^37.14N  LONG=71^20.83W  
PBEACON  DELAY=21  EVERY=30                         LAT=42^37.14N  LONG=71^20.83W

closed 2022-05-24 by wb2osz

Tyler-2, 2022-05-24:
That's actually where I started and it got complicated in a hurry with multiple OBEACONs in the mix.

In the example you've shared there's a 30 second rotation where the first 10 seconds is a WIDE, the second 10 is a DIRECT, and the last 10 is a DIRECT. That works.

0:31 seconds we do wide1-1.
0:41 seconds we do direct.
0:51 seconds we do direct.
1:01 seconds we do wide1-1.
1:11 seconds we do direct.
1:21 seconds we do direct.
etc

So to accomplish this:
```
- Every minute the packet goes DIRECT path (no digi)
- Every 2nd minute it goes via WIDE1-1 path
- Every 4th minute it goes via WIDE1-1,WIDE2-1
- Every 8th minute it goes via WIDE1-1,WIDE2-2 (only if 3 hops is OK)
```

1. DELAY=0:01 EVERY=8:00 VIA=DIRECT
2. DELAY=1:01 EVERY=8:00 VIA=WIDE1-1
3. DELAY=2:01 EVERY=8:00 VIA=DIRECT
4. DELAY=3:01 EVERY=8:00 VIA=WIDE1-1,WIDE2-1
5. DELAY=4:01 EVERY=8:00 VIA=DIRECT
6. DELAY=5:01 EVERY=8:00 VIA=WIDE1-1
7. DELAY=6:01 EVERY=8:00 VIA=DIRECT
8. DELAY=7:01 EVERY=8:00 VIA=WIDE1-1,WIDE2-2

For something like a repeater obeacon with net times and stuff on it this becomes a lot of lines.

If you have multiple obeacons, i assume you should space them out, so you would shift the `DELAY` for each OBEACON.

I don't know how I would construct a "cleaner" way of specifying this while maintaining flexibility, so I guess I'll just leave all this here as a reference for future generations of mankind.

Tyler-2, 2022-05-25:
```
#W4FEG
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=0:01 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=1:01 via=WIDE1-1
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=2:01 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=3:01 via=WIDE1-1,WIDE2-1
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=4:01 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=5:01 VIA=WIDE1-1
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=6:01 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.794143008 long=-77.93201899 height=350 objname=444.650ml power=50 gain=6 alt=100 SYMBOL=/r FREQ=444.650 OFFSET=+5 TONE=203.5 COMMENT="W4FEG" EVERY=8:00 DELAY=7:01 VIA=WIDE1-1,WIDE2-2

#N4POW
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=0:06 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=1:06 VIA=WIDE1-1
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=2:06 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=3:06 VIA=WIDE1-1,WIDE2-1
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=4:06 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=5:06 VIA=WIDE1-1
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=6:06 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.54937100 long=-77.92727900 alt=115 objname=147.315ml power=50 height=120 gain=6 SYMBOL=/r FREQ=147.315 OFFSET=+.6 TONE=74.4 COMMENT="Net M 7PM Mtg3rdTu N4POW" EVERY=8:00 DELAY=7:06 VIA=WIDE1-1,WIDE2-2

#WB4IKL
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=0:12 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=1:12 VIA=WIDE1-1
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=2:12 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=3:12 VIA=WIDE1-1,WIDE2-1
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=4:12 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=5:12 VIA=WIDE1-1
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=6:12 VIA=DIRECT
OBEACON sendto=0,IG compress=0 lat=37.793540 long=-77.930220 alt=105 objname=147.270ml power=50 height=200 gain=6 SYMBOL=/r FREQ=147.270 OFFSET=+.6 TONE=203.5 COMMENT="Net M 8PM WB4IKL" EVERY=8:00 DELAY=7:12 VIA=WIDE1-1,WIDE2-2
```

Now I get a warning about exceeding the maximum number of beacons, when really, these are just three beacons set up to be friendlier to the RF network than is typical around here.

wb2osz, 2022-05-25:
Your examples have twice the number needed:

EVERY=8:00 DELAY=0:01 VIA=DIRECT
EVERY=8:00 DELAY=1:01 via=WIDE1-1
EVERY=8:00 DELAY=2:01 VIA=DIRECT
EVERY=8:00 DELAY=3:01 via=WIDE1-1,WIDE2-1

EVERY=8:00 DELAY=4:01 VIA=DIRECT
EVERY=8:00 DELAY=5:01 VIA=WIDE1-1
EVERY=8:00 DELAY=6:01 VIA=DIRECT
EVERY=8:00 DELAY=7:01 VIA=WIDE1-1,WIDE2-2

You could use half as many, with a 4 minute cycle time, and get the same result.

It's not as simple as turning on some option, to get some fixed set of paths, but this is very flexible.
Suggestions for improvement are always welcome.