wb2osz/direwolf #397: docs question: HEIGHT parameter

Tyler-2, 2022-05-25:
HEIGHT: Antenna height in feet

Is this height above ground, height above sea level, or height above average terrain? Unqualified I think most people would just put height above ground. But this is used to generate PHG parameters and PHG uses HAAT.

I suspect it is HAAT, but I'd like clarification in the docs to confirm.

dranch, 2022-05-25:
This isn't a Direwolf thing per se but an APRS thing.  Per http://www.aprs.org/phg.html :
--
PHG stands for Power-Height-Gain (and directivity) which is all you need to calculate the relative RF range of a station. The most important parameter being Height Above Average Terrain. Because the FCC has a rigorous method of mathematically calculating, some poeople think that HAAT takes on some exact, almost mystical meaning. When in fact, it is just the simple Height of your antenna over the surrounding area that it serves. Notice that Height above sealevel means absolutely nothing. It is your heigh over everything around you that is important. 
--

wb2osz, 2023-05-19:
Thank you for pointing this out.
The (not yet released) next User Guide clarifies this.

73,
John WB2OSZ

closed 2023-05-19 by wb2osz