wb2osz/direwolf #399: Power Control Feature Request

N7FMH, 2022-06-02:
Greetings,

I would like to request a “Power Control” feature.

Reason: to conserve battery power on stations that may be primarily transmit only (E.g. weather station).

Principle feature:
1.	Apply power to a radio
2.	Wait x seconds 
3.	Transmit
4.	Shutdown radio

Secondary feature:
1.	Apply power to radio to listen at prescribed times for possible incoming commands

Thank you!

Best regards,
Fred N7FMH

dranch, 2022-06-02:
All of that can be done today outside of Direwolf using tools like cron, GPIO pins on a Raspberry Pi / parallel port on a PC etc to control a relay powering up/down a radio, etc.

N7FMH, 2022-06-03:
Thank you for your response and pointing me towards another option.

Inside Direwolf:
Advantages: It is a simple to use parameter that can be used by people without scripting(?) or other skill sets.
Disadvantage: The Pi/Direwolf is always running and using the most power. Someone spends time adding/debugging the feature amidst all the other feature requests.

Outside Direwolf:
Advantages: No impact on Direwolf WB2OSZ/contributors. In theory, without Direwolf running, the power consumption would be at its lowest between transmit periods. [Note: I am using a RPi Zero.]
Disadvantage:
Personally, this would be new ground for me as I suspect it would be for others.
I have used cron. Can't say I know it.
I have never written a script (equivalent to a "batch" file. There are several examples of 'sh' files so maybe it is straight forward.
I do know that I can set/reset GPIO pins from the command line.
Starting Direwolf is easy, killing it is easy (assuming it can be killed from within the script), knowing 'when' is another thing. Sending a single packet VS. multiple packets over a minute.

I do appreciate all the time and effort it takes to create a masterpiece like Direwolf. I myself work in an R&D environment and nothing happens quickly or without a lot of effort/money. I understand that if this was added to a "to do" list, it might be a very low priority.  I believe I am the only one to ask about this feature in about 6 years so it probably makes it a non-starter. Did I just kill it myself? However, without asking it would never be a possibility.

Thank you!

dranch, 2022-06-03:
I will send you an email to give you an example of what I do for a similar need and if that meets your needs.