wb2osz/direwolf #401: RX of TX frames (local echo) caused by cross talk introduced by TRRS plugs and cables used with USB sound devices

HarlekinSimplex, 2022-06-15:
Actually I see that sent beacons or KISS frames are received as local achoes and recognized and logged during TX even if DW operates in simplex (FULLDUP OFF) mode.

The root cause of course is a hardware issue that introduces a faint cross talk by using TRRS cables with joint groundings of playback and capture to connect a USB Interface with radios that don't have build in USB Codecs and/or CAT interfaces.
As a typical device of that art there is digiri mobil or like DIY interfaces with our without GPIO or VOX triggered PTT.
As soon as you use a TRRS Plug and cable of more than a couple cm of length, such a faint cross talk appears on the input line while sending data.

Though a hardware sulution may be possible on the interface side, avoiding such faint echoes on standard TRRS cables entirely is more or less impossible.

One possible software soultion, at least for using dw in simplex mode, would be to just ignore any incomming data while sending.
How ever, it may be difficult to handle the possible delay between TX and RX data to distinguish between echoes and real RX data.

Another possible solution would be to compare incomming data with the actual sent data and as soon RX frames appear that resemble TX frames that have been recently sent (within a very short time) these echo frames may be just discarded.

Since the cross talk comes in with a rather low audio level (approx. audio level 0..5) having an option to define a minimum audio level for processing incoming singnals would solve this crosstalk issue not only for simplex but for duplex and fullduplex as well.

This local echoes seem not be a problem with other software TNC, at least in simplex mode.
At least I did not find anything regarding this on the web so far and friends of mine did not report such issues with soundmodem or like as well.
Same goes with dw, nothing regarding such an issue is to be found on the internet by now.
I have no clue why that is so. Maybe the consequences of these echoes are not causing any major issus with the majority of applications. In my case I try to run the RNS (Reticulum) stack wit KISS frames and dw in simplexe mode and those local echoes seem to disrupt the communication antirely. Having set up to USB sound device resembling two fully sepatated audio paths (ADEVICE with separated Input and Output devices) did not produce any echoes enymore and the RNS KISS protocol immediately did run perfect without any issues. So these echoes are clearly a show stopper for me and a solution need to be found even if other applications may be more resiliant to such inbound copies of outbound frames.

How ever the problem can be reproduced quite easily.
Just set up a 10sec beacon like this:
OBEACON comment="Test Beacon" delay=0:10 every=00:10 OBJNAME=Test lat=0^0.0 long=0^0.0 dest=CQ

Plug in any USB Sound device you like with a TRRS socket attached to connect to the radio with and you'll see the local echoes comming in, regardless if you have connect anything to the TRRS socket already or not.

I tested this with several brands of such USB devices and the results were always the same.
Local echos at an audio level around 0..5 max. properly decoded and flawlessly reported to the console by dw as incomming data.
The longer the cables the higer the audio level gets.
You even may use these echoes as a logging feature to see that audio encoding and decoding with dw does work and all sent frames are 100% error free received, decoded and processed again. And I thought that was intentionally implemented this way until I muted the campture channel using alsamixed during my investigation why the RNS stack does not work and the 'logging' stopped immediately ... !

The echoes could only be subdued by lowering the Playback and Capture levels to ridiculously low levels which usually leads to an sitution where the communication gets unacceptable unreliable just because of to low audio levels.

I talked already to the manufacturer of digirig mobile and he was very supportive.
However the interface HW uses TRRS plugs and the echoes do _only_ appear if a TRRS plug with a cable is inserted.
So the issue is primarily introduced by the cable and the joint usage of mic and speaker grounding in TRRS plug and cable.
How ever, my experiences shows that ANY cable, regardless how it is build, introduces those echoes as soon as the TRRS Plug is set into the socket. Even separated shielding of mic and speaker right after the plug did not reduce the cross talk entirely.
Only cutting the joint grounding appart on the PCB to get 100% separated audio paths from the decoupling signal transformes to the radio together with proper shielding of both audio paths will stop the cross talk.
So more or less each and every USB sound device using TRRS seem to be prone to that issue.

Since there are thousend like interfaces already in the wild, solving that for actual HW needs a software adoption like I have suggest above.

Would it be possible to implement one of those suggested soultions shortly?
My whish would be to have the audio level filter implemented to ignore frames below a certain audio level.
I guess that one would be the most simple, easy to implement and fexible solution (more or less one if statement + plus the new commandline switch and config option) to cover any crosstalk issues like that with TRRS plugs and cables.

pls adv
Cheers
Stephan de DD8SB

na7q, 2022-09-08:
I will add that this issue is common. Rarely problematic. Certainly annoying. Surprisingly never hear much of it either.

Tyler-2, 2022-09-23:
I had been seeing this too and thought it was just a "logging" feature like you said. But then I added a filter to not digipeat my own packets, which seemed like a strange thing to have to do.

marrold, 2022-12-29:
I have been experimenting with "virtualised" Dire Wolf instances with audio loopbacks and experience the same issue. It makes sense as there's no PTT or radio switching from Rx to Tx, so Dire Wolf can always "hear" what's on the loopback.

It would be great if these packets could be filtered from the logs / KISS / AGW etc. For digipeating or igating it's less of an issue as these can be filtered already.

Thanks

marrold, 2022-12-29:
Looking at the issues, these all seem related:

#401
#97
#288
#337

wb2osz, 2023-05-08:
Fixed in the dev branch and will be in 1.7 release. 
Received audio is muted while transmitting.

closed 2023-05-08 by wb2osz