wb2osz/direwolf #402: Build error against libgps 3.24

yuvadm, 2022-06-19:
Similar to #357 build fails when building against libgps 3.24

```
[ 37%] Building C object src/CMakeFiles/direwolf.dir/dwgpsd.c.o
src/direwolf/src/dwgpsd.c:68:2: error: #error libgps API version might be incompatible.
   68 | #error libgps API version might be incompatible.
      |  ^~~~~
make[2]: *** [src/CMakeFiles/direwolf.dir/build.make:1014: src/CMakeFiles/direwolf.dir/dwgpsd.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:538: src/CMakeFiles/direwolf.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

wb2osz, 2022-06-20:
Which version of direwolf is this?
Try using the "dev" branch.

yuvadm, 2022-06-24:
@wb2osz I am already on `dev` branch, head commit `d11bd6f`

meltdown03, 2022-06-29:
I changed this line to a `#warning` instead of an error and rebuilt it.
https://github.com/wb2osz/direwolf/blob/7d3c1d100ea38605bf5d11459f5a678be99fffab/src/dwgpsd.c#L64

yuvadm, 2022-06-29:
@meltdown03 does the build work properly? No breaking changes in gpsd?

meltdown03, 2022-06-29:
> Sometimes a major API number change includes incompatibilities with
> previous versions.
> It might compile successfully but not operate properly without additional
> source code changes.
> You will find conditional compilation, in several places, dealing with
> those incompatible API changes.
> Did you test for correct operation?

No, I didn't have a chance to test it yet.

yuvadm, 2022-10-05:
There's a comment on Arch Linux AUR that confirms no regression with new gpsd versions https://aur.archlinux.org/packages/direwolf-git#comment-872500

wb2osz, 2023-04-10:
This has been tested on gpsd versions up through 3.25.  Closing.

closed 2023-04-10 by wb2osz