wb2osz/direwolf #403: Kissutil not being installed from apt

Frog-Lover-Antony, 2022-06-20:
Running `sudo apt install direwolf` on fresh install of linux mint (ubuntu-based) results in all packages being built and installed except for kissutil - it is built but not installed.
https://www.mail-archive.com/debian-bugs-dist@lists.debian.org/msg1733634.html

dranch, 2022-06-20:
That original report reflects Direwolf v1.5 which is very old.  Direwolf v1.6 has this utility already included so you need to get Linux Mint to upgrade to a newer version of Direwolf to get it included.  Alternatively, you can compile and package your own version of Direwolf.  If you agree, please close this Github issue.

wb2osz, 2022-06-20:
Version 1.5 is obsolete.  Use Version 1.6 which was released in October 2020.

closed 2022-06-20 by wb2osz