wb2osz/direwolf #405: Messages gated from APRSIS to RF are stripped off their sequence number

F4FXL, 2022-07-01:
_**Steps to reproduce:**_
Send a message from an APRSIS only client to an RF only client. E.g. I used my APRSDroid to send a message to my FT2.

_**Expected Result:**_
The message goes onto RF unchanged and with its sequence number

_**Actual Result**_
The sequence number is removed from the message when going on air. This will prevent the recipient to send any ack packet because it did not get the sequence number. Additionally it will lead to unnecessary traffic as the sender ill continue sending the message as long as it does not get the ack.

See topic here : [https://groups.io/g/direwolf/topic/92085159](https://groups.io/g/direwolf/topic/92085159)

wb2osz, 2022-07-04:
Thank you for providing a clear complete description including an example to demonstrate the problem.
This has been fixed in the dev branch.

closed 2022-07-04 by wb2osz

wb2osz, 2023-01-28:
commit 30869c7afc5517eaf9f5a31bd21274215a4c5225
Date:   Mon Jul 4 02:11:55 2022 +0100

    Issues 405 & 406 and other improvements in message interpretation.

This is in the "dev" branch which will become release 1.7.